Klipper3d/klipper #5211: [heater_fan my_nozzle_fan] dual nozzle set

gong-Gs, 2022-01-30:
One cooling fan was used for the dual nozzle heat sink, and it was set as follows.
The problem is that the fan operates when the temperature of nozzle0 is above 50°C, and the fan does not operate even if the temperature of nozzle1 is above 50°C. I'm curious about the way.
It seems that the parameter extruder1 is not recognized.
Thank you...

[heater_fan my_nozzle_fan]
pin: PE6
max_power: 1.0
shutdown_speed: 0
#cycle_time:
#kick_start_time:
#off_below:
heater: extruder ,extruder1
#heater: extruder1
heater_temp: 50.0
fan_speed: 0.5

Sineos, 2022-01-30:
Does it change if you remove the blank, i.e.
`heater: extruder ,extruder1`
to
`heater: extruder,extruder1`
or
`heater: extruder, extruder1`

gong-Gs, 2022-01-31:
> Does it change if you remove the blank, i.e. `heater: extruder ,extruder1` to `heater: extruder,extruder1` or `heater: extruder, extruder1`

Thank you for your answer, Sineos.
I just tried it, and it's the same symptoms...
If you add the number (1 or 2) after the extruder, the two letters "er1" turn blue and it doesn't recognize...

DrumClock, 2022-02-14:
try this:

```
[duplicate_pin_override]
pins: PE6

[heater_fan my_nozzle_fan]
pin: PE6
max_power: 1.0
shutdown_speed: 0
#cycle_time:
#kick_start_time:
#off_below:
heater: extruder 
heater_temp: 50.0
fan_speed: 0.5

[heater_fan my_nozzle_fan1]
pin: PE6
max_power: 1.0
shutdown_speed: 0
#cycle_time:
#kick_start_time:
#off_below:
heater: extruder1
heater_temp: 50.0
fan_speed: 0.5
```

DrumClock, 2022-02-14:
so how to solve a common FAN for two extruders?

DrumClock, 2022-02-14:
**now I've tried this:**

```
[heater_fan fanE0]
pin: PC8
heater: extruder, extruder1
heater_temp: 30.0
fan_speed: 1.0
```

**it's okay for me .....**  Firmware version: v0.10.0-245-ga7b01857-dirty

Sineos, 2022-02-18:
Topic seems to be concluded. For further discussions please do not hesitate to open a new topic on https://klipper.discourse.group/.

closed 2022-02-18 by Sineos