Klipper3d/klipper #5212: Unable to parse move 'G1 E- F2100'

1stsetup, 2022-01-30:
Multiple times I get the following error and printing halts on the same gcode file: Unable to parse move 'G1 E- F2100'

[klippy.log.zip](https://github.com/Klipper3d/klipper/files/7965973/klippy.log.zip)

Sineos, 2022-01-30:
This is caused by your `[gcode_macro PAUSE]`, which seems wrong / incomplete.

If you need further assistance, please open a topic on https://klipper.discourse.group

1stsetup, 2022-01-30:
I  found the problem that the variable E was never set in the gcode_macro PAUSE code when executing "G1 E-{E} F2100" when my filament runout sensor triggered.

It would be nice to have an error message stating that the variable E is not set/undefined/empty next to this rather cryptic message.

closed 2022-01-30 by 1stsetup