Klipper3d/klipper #5215: error: Internal error in MCU 'mcu' stepcompress

Jamin82, 2022-01-31:
Getting the occasional error on G1 moves randomly:

error: Internal error in stepcompress
Exiting SD card print (position 543961)
Stats 175665.0: gcodein=0 mcu: mcu_awake=0.017 mcu_task_avg=0.000015 mcu_task_stddev=0.000031 bytes_write=969516 bytes_read=423308 bytes_retransmit=9 bytes_invalid=0 send_seq=24518 receive_seq=24518 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=9 stalled_bytes=22763 freq=167997929  raspberry_pi: temp=51.1 heater_bed: target=65 temp=64.8 pwm=0.192 extruder_motor: temp=34.1 sysload=0.13 cputime=54.490 memavail=1666200 print_time=849.241 buffer_time=2.326 print_stall=0 extruder: target=215 temp=215.1 pwm=0.240
Dumping stepper 'stepper_z' (mcu) 1 queue_step:
queue_step 0: t=142283444422 p=-95575 i=29654995 c=1 a=0
Dumping stepper 'stepper_y' (mcu) 32 queue_step:
queue_step 0: t=142279560162 p=73980 i=61048522 c=-1 a=0
queue_step 1: t=142279779786 p=73979 i=219624 c=-1 a=0
queue_step 2: t=142279903603 p=73978 i=123817 c=-8 a=0
queue_step 3: t=142280869100 p=73970 i=98778 c=-1 a=0
queue_step 4: t=142280925161 p=73969 i=56061 c=-15 a=0
queue_step 5: t=142281769939 p=73954 i=59924 c=-2 a=18668
queue_step 6: t=142281915933 p=73952 i=67402 c=-4 a=0
queue_step 7: t=142282189871 p=73948 i=71732 c=-2 a=18680
queue_step 8: t=142282362105 p=73946 i=81822 c=-3 a=-7584
queue_step 9: t=142282573691 p=73943 i=70694 c=-10 a=0
queue_step 10: t=142283411205 p=73933 i=201268 c=-1 a=0
queue_step 11: t=142283540054 p=73932 i=128849 c=-3 a=-4822
queue_step 12: t=142283885958 p=73929 i=102672 c=-2 a=-14001
queue_step 13: t=142284060675 p=73927 i=86046 c=-3 a=0
queue_step 14: t=142284330753 p=73924 i=97986 c=-5 a=2668
queue_step 15: t=142284896625 p=73919 i=147248 c=-1 a=0
queue_step 16: t=142284965624 p=73918 i=68999 c=-4 a=-2232
queue_step 17: t=142285221849 p=73914 i=62620 c=-2 a=23828
queue_step 18: t=142285392120 p=73912 i=83823 c=-4 a=0
queue_step 19: t=142285756131 p=73908 i=112542 c=-3 a=-1770
queue_step 20: t=142286068638 p=73905 i=92733 c=-5 a=0
queue_step 21: t=142286514977 p=73900 i=75407 c=-3 a=0
queue_step 22: t=142286808373 p=73897 i=142582 c=-1 a=0
queue_step 23: t=142287005937 p=73896 i=197564 c=-3 a=0
queue_step 24: t=142287577896 p=73893 i=176831 c=-2 a=-28782
queue_step 25: t=142287874094 p=73891 i=148149 c=-2 a=-4030
queue_step 26: t=142288127019 p=73889 i=108806 c=-2 a=-14588
queue_step 27: t=142288335031 p=73887 i=113794 c=-2 a=9527
queue_step 28: t=142288554483 p=73885 i=96131 c=-4 a=1150
queue_step 29: t=142288958242 p=73881 i=108466 c=-4 a=14038
queue_step 30: t=142289543959 p=73877 i=176091 c=-1 a=0
queue_step 31: t=142289622410 p=73876 i=78451 c=-3 a=-3269
Dumping stepper 'stepper_x' (mcu) 13 queue_step:
queue_step 0: t=142218780746 p=31012 i=9281 c=-6532 a=0
queue_step 1: t=142279404769 p=24480 i=9812 c=-111 a=-6
queue_step 2: t=142280456842 p=24369 i=9383 c=-149 a=0
queue_step 3: t=142281854263 p=24220 i=8737 c=-115 a=8
queue_step 4: t=142282911991 p=24105 i=9270 c=-110 a=5
queue_step 5: t=142283961770 p=23995 i=9374 c=-96 a=3
queue_step 6: t=142284875132 p=23899 i=9152 c=-53 a=4
queue_step 7: t=142285365539 p=23846 i=8991 c=-92 a=4
queue_step 8: t=142286209554 p=23754 i=9090 c=-93 a=5
queue_step 9: t=142287076343 p=23661 i=9119 c=-106 a=8
queue_step 10: t=142288089122 p=23555 i=10764 c=-78 a=7
queue_step 11: t=142288950701 p=23477 i=11730 c=-49 a=86
queue_step 12: t=142289632025 p=23428 i=17148 c=-21 a=324
Dumping stepper 'stepper_z2' (mcu) 1 queue_step:
queue_step 0: t=142283444422 p=-95594 i=29654995 c=1 a=0
Dumping stepper 'stepper_z1' (mcu) 1 queue_step:
queue_step 0: t=142283444422 p=-95741 i=29654995 c=1 a=0
Dumping stepper 'extruder' (mcu) 36 queue_step:
queue_step 0: t=142217145571 p=270697 i=96417 c=474 a=0
queue_step 1: t=142262846453 p=271171 i=95641 c=6 a=439
queue_step 2: t=142263427218 p=271177 i=95975 c=22 a=45
queue_step 3: t=142265548198 p=271199 i=95110 c=8 a=380
queue_step 4: t=142266318075 p=271207 i=93467 c=4 a=2165
queue_step 5: t=142266704159 p=271211 i=92693 c=6 a=1568
queue_step 6: t=142267285688 p=271217 i=94544 c=9 a=412
queue_step 7: t=142268152968 p=271226 i=96096 c=7 a=-169
queue_step 8: t=142268823445 p=271233 i=97450 c=10 a=-159
queue_step 9: t=142269792137 p=271243 i=98797 c=3 a=-2645
queue_step 10: t=142270078468 p=271246 i=96672 c=11 a=0
queue_step 11: t=142271139513 p=271257 i=94325 c=7 a=700
queue_step 12: t=142271808071 p=271264 i=87908 c=2 a=17619
queue_step 13: t=142272009662 p=271266 i=96064 c=8 a=0
queue_step 14: t=142272782584 p=271274 i=100474 c=2 a=-6089
queue_step 15: t=142272972922 p=271276 i=95953 c=6 a=0
queue_step 16: t=142273553004 p=271282 i=100317 c=2 a=-7275
queue_step 17: t=142273740797 p=271284 i=94751 c=4 a=1854
queue_step 18: t=142274129953 p=271288 i=93779 c=7 a=810
queue_step 19: t=142274802809 p=271295 i=93172 c=5 a=1595
queue_step 20: t=142275284127 p=271300 i=92680 c=4 a=3129
queue_step 21: t=142275673646 p=271304 i=92705 c=6 a=1379
queue_step 22: t=142276249632 p=271310 i=91776 c=4 a=3424
queue_step 23: t=142276637862 p=271314 i=92358 c=5 a=1569
queue_step 24: t=142277113030 p=271319 i=90046 c=2 a=14698
queue_step 25: t=142277311734 p=271321 i=93960 c=5 a=999
queue_step 26: t=142277793576 p=271326 i=96012 c=6 a=0
queue_step 27: t=142278371300 p=271332 i=97664 c=4 a=475
queue_step 28: t=142278759435 p=271336 i=92293 c=6 a=1351
queue_step 29: t=142279337039 p=271342 i=95874 c=5 a=0
queue_step 30: t=142279818217 p=271347 i=97682 c=3 a=-1960
queue_step 31: t=142280103608 p=271350 i=95907 c=4 a=1711
queue_step 32: t=142280493780 p=271354 i=92185 c=4 a=2716
queue_step 33: t=142280882914 p=271358 i=96283 c=81 a=3
queue_step 34: t=142288873242 p=271439 i=277968 c=-1 a=0
queue_step 35: t=142288970727 p=271438 i=97485 c=-12 a=-2625
Dumping trapq 'toolhead' 7 moves:
move 0: pt=846.529993 mt=0.378691 sv=80.000000 a=0.000000 sp=(320.674983,82.503983,0.264266) ar=(-0.707107,-0.707107,0.000092)
move 1: pt=846.908684 mt=0.014146 sv=80.000000 a=0.000000 sp=(299.253000,61.082000,0.267051) ar=(-0.839452,-0.543434,0.000032)
move 2: pt=846.922830 mt=0.010050 sv=80.000000 a=0.000000 sp=(298.303000,60.467000,0.267087) ar=(-0.702740,-0.711447,0.000193)
move 3: pt=846.932880 mt=0.010633 sv=80.000000 a=0.000000 sp=(297.738000,59.895000,0.267243) ar=(-0.826401,-0.563081,0.000050)
move 4: pt=846.943514 mt=0.013198 sv=80.000000 a=0.000000 sp=(297.035000,59.416000,0.267285) ar=(-0.714281,-0.699859,0.000181)
move 5: pt=846.956712 mt=0.003816 sv=80.000000 a=-6000.000000 sp=(296.280835,58.677063,0.267477) ar=(-0.714281,-0.699859,0.000181)
move 6: pt=846.960527 mt=0.009081 sv=57.106469 a=-6000.000000 sp=(296.094000,58.494000,0.267524) ar=(-0.877614,-0.479369,-0.000024)
Dumping trapq 'extruder' 7 moves:
move 0: pt=846.529993 mt=0.378691 sv=2.994689 a=0.000000 sp=(465.184559,0.000499,0.000000) ar=(1.000000,0.025000,0.000000)
move 1: pt=846.908684 mt=0.014146 sv=2.992336 a=0.000000 sp=(466.318620,0.000000,0.000000) ar=(1.000000,0.025000,0.000000)
move 2: pt=846.922830 mt=0.010050 sv=2.995041 a=0.000000 sp=(466.360950,0.000000,0.000000) ar=(1.000000,0.025000,0.000000)
move 3: pt=846.932880 mt=0.010633 sv=2.995264 a=0.000000 sp=(466.391050,0.000000,0.000000) ar=(1.000000,0.025000,0.000000)
move 4: pt=846.943514 mt=0.013198 sv=2.994364 a=0.000000 sp=(466.422900,0.000000,0.000000) ar=(1.000000,0.025000,0.000000)
move 5: pt=846.956712 mt=0.003816 sv=2.994364 a=-224.577273 sp=(466.462420,0.000988,0.000000) ar=(1.000000,0.025000,0.000000)
move 6: pt=846.960527 mt=0.009081 sv=2.141572 a=-225.008321 sp=(466.472210,0.000000,0.000000) ar=(1.000000,0.025000,0.000000)
Requested toolhead position at shutdown time 846.863380: (301.81580413644343, 63.64480413644341, 0.26671809328371604)
Stats 175666.0: gcodein=0 mcu: mcu_awake=0.017 mcu_task_avg=0.000015 mcu_task_stddev=0.000031 bytes_write=971853 bytes_read=427233 bytes_retransmit=9 bytes_invalid=0 send_seq=24557 receive_seq=24557 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=32 stalled_bytes=20656 freq=167997941  raspberry_pi: temp=51.6 heater_bed: target=65 temp=64.7 pwm=0.244 extruder_motor: temp=34.2 sysload=0.13 cputime=54.544 memavail=1666204 print_time=849.241 buffer_time=1.325 print_stall=0 extruder: target=215 temp=214.9 pwm=0.240
stepcompress o=18 i=0 c=2 a=0: Invalid sequence
Exception in flush_handler
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/toolhead.py", line 392, in _flush_handler
    self.flush_step_generation()
  File "/home/pi/klipper/klippy/toolhead.py", line 346, in flush_step_generation
    self._update_move_time(max(self.print_time, self.last_kin_flush_time))
  File "/home/pi/klipper/klippy/toolhead.py", line 292, in _update_move_time
    m.flush_moves(mcu_flush_time)
  File "/home/pi/klipper/klippy/mcu.py", line 939, in flush_moves
    % (self._name,))
error: Internal error in MCU 'mcu' stepcompress
[klippy (1).log](https://github.com/Klipper3d/klipper/files/7970822/klippy.1.log)

Jamin82, 2022-02-01:
re flashed fluidd and klipper on fresh sd card and loaded configs. problem is still cropping up unfortunately.
this time when i was trying to unload filament with my lcd screen
[klippy (2).log](https://github.com/Klipper3d/klipper/files/7977802/klippy.2.log)

KenK73, 2022-02-06:
> Exception in flush_handler
> Traceback (most recent call last):
>   File "/home/pi/klipper/klippy/toolhead.py", line 392, in _flush_handler
>     self.flush_step_generation()
>   File "/home/pi/klipper/klippy/toolhead.py", line 337, in flush_step_generation
>     self.move_queue.flush()
>   File "/home/pi/klipper/klippy/toolhead.py", line 175, in flush
>     self.toolhead._process_moves(queue[:flush_count])
>   File "/home/pi/klipper/klippy/toolhead.py", line 333, in _process_moves
>     self._update_move_time(next_move_time)
>   File "/home/pi/klipper/klippy/toolhead.py", line 292, in _update_move_time
>     m.flush_moves(mcu_flush_time)
>   File "/home/pi/klipper/klippy/mcu.py", line 939, in flush_moves
>     % (self._name,))
> error: Internal error in MCU 'mcu' stepcompress

Just upgraded Klipper on my vCore3 and I cant even get a print started. Trying to print a calibration cube that printed fine before the upgrade.

dallerman, 2022-02-07:
Samme issue here started happening today. I have a BTT SKR mini e3 V3 been running it for about a week and suddenly reporting Internal error on command:"G1" and just stop printing
[klippy (23).log](https://github.com/Klipper3d/klipper/files/8018040/klippy.23.log)

I have tested another PSU as i did some fan mods on it some days ago.
I have re-seated all cables and looked for shorts but have not been able to pinpoint any at this time

dallerman, 2022-02-08:
After some additional troubleshooting i found  the following lines in the log

TMC 'extruder' reports DRV_STATUS: 40140090 s2vsa=1(LowSideShort_A!) olb=1(OpenLoad_B!) cs_actual=20 stealth=1
Transition to shutdown state: TMC 'extruder' reports error: DRV_STATUS: 40140090 s2vsa=1(LowSideShort_A!) olb=1(OpenLoad_B!) cs_actual=20 stealth=1
****

as i have 2209 drivers they were configured to use stealthchop_threshold: 999999 i deisabled by stealthchop_threshold: 0 and now it seems to print fine

mistrjirka, 2022-02-12:
Hi, 
I have the same issue,  I have 2208 drivers in simple step mode (so I didn´t configure anything about the drivers) and I have the same issue on this gcode: 
[GASB1_V7_canopy_0.15mm_PLA__3h7m.gcode.log](https://github.com/Klipper3d/klipper/files/8054208/GASB1_V7_canopy_0.15mm_PLA__3h7m.gcode.log)
I will try to print it again and find out if it is consistent
[klippy-1.log](https://github.com/Klipper3d/klipper/files/8054205/klippy-1.log)

KenK73, 2022-03-02:
I backed up my printer.cfg, bought new SanDisk 8GB cards for pi and SKR, wiped everything, installed latest RatOS, then did the online updates.  Everything working fine since.

Sineos, 2022-03-02:
General note: Log lines like `stepcompress o=4 i=0 c=17 a=0: Invalid sequence` often are a consequence of errors / events that happened before. This means they are rather a symptom but not the root cause.

Carefully check your logs for other errors or noteworthy events. In doubt or if support is needed open a topic on https://klipper.discourse.group/

Sineos, 2022-03-25:
Closing since it seems to have concluded. Follow ups please on Discourse

closed 2022-03-25 by Sineos