Klipper3d/klipper #5216: communications dropouts

tommorox2345, 2022-01-31:
so i have been enjoing klipper for a few months now with no issues whatsoever and all of a sudden i get the issue of printing fine until it just randomly looses communication with the mcu. creality 4.2.7 board and a pi 4 4gb. the text below is 2 snippets of the log when it goes wrong

i have heard it can be power related but i am running the official pi power supply so i dont believe thats the case

things i have tried:
new cable
increased pi cooling
increased mainboard cooling
restarted everything

im all ears if anyone has a suggestion

Stats 96219.8: gcodein=456845 mcu: mcu_awake=0.021 mcu_task_avg=0.000036 mcu_task_stddev=0.000033 bytes_write=1295804 bytes_read=195540 bytes_retransmit=9 bytes_invalid=0 send_seq=22755 receive_seq=22755 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=11609 freq=72000291 heater_bed: target=110 temp=110.0 pwm=0.627 sysload=0.26 cputime=1719.914 memavail=3103360 print_time=753.796 buffer_time=0.000 print_stall=0 extruder: target=260 temp=259.8 pwm=0.591
Stats 96220.8: gcodein=456845 mcu: mcu_awake=0.021 mcu_task_avg=0.000036 mcu_task_stddev=0.000033 bytes_write=1295804 bytes_read=195540 bytes_retransmit=9 bytes_invalid=0 send_seq=22755 receive_seq=22755 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=11610 freq=72000291 heater_bed: target=110 temp=110.0 pwm=0.627 sysload=0.26 cputime=1719.916 memavail=3102580 print_time=753.796 buffer_time=0.000 print_stall=0 extruder: target=260 temp=259.8 pwm=0.591
Stats 96221.8: gcodein=456901 mcu: mcu_awake=0.021 mcu_task_avg=0.000036 mcu_task_stddev=0.000033 bytes_write=1295804 bytes_read=195540 bytes_retransmit=9 bytes_invalid=0 send_seq=22755 receive_seq=22755 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=11611 freq=72000291 heater_bed: target=110 temp=110.0 pwm=0.627 sysload=0.26 cputime=1719.919 memavail=3102964 print_time=753.796 buffer_time=0.000 print_stall=0 extruder: target=260 temp=259.8 pwm=0.591
Lost communication with MCU 'mcu'
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

Read 95780.181291: 'N12514 G1 X140.662 Y49.256 E0.36785*85\n'
Read 95780.183910: 'N12515 G1 X142.259 Y50.854 E0.44478*94\n'
Read 95780.186527: 'N12516 G1 X142.259 Y51.45 E0.46506*110\n'
Read 95780.189127: 'N12517 G1 X140.277 Y49.467 E0.56055*94\n'
Read 95780.191773: 'N12518 G1 X139.851 Y49.636 E0.57616*87\n'
Read 95780.194961: 'N12519 G1 X142.26 Y52.045 E0.69216*99\n'
Read 95780.198528: 'N12520 G1 X142.26 Y52.641 E0.71244*101\n'
Read 95780.201254: 'N12521 G1 X139.415 Y49.796 E0.84945*93\n'
Read 95780.203865: 'N12522 G1 X138.903 Y49.88 E0.8671*81\n'
Read 95780.206470: 'N12523 G1 X142.26 Y53.237 E1.02875*111\n'
gcode state: absolute_coord=True absolute_extrude=True base_position=[0.0, 0.0, 0.0, 742.7718800000002] last_position=[142.26, 53.237, 1.0, 743.8006300000002] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.0166666666667 extrude_factor=1.0 speed=130.267235
Reactor garbage collection: (95408.422541298, 0.0, 0.0)
Stats 95780.3: gcodein=445395 mcu: mcu_awake=0.021 mcu_task_avg=0.000036 mcu_task_stddev=0.000033 bytes_write=1295804 bytes_read=195540 bytes_retransmit=9 bytes_invalid=0 send_seq=22755 receive_seq=22755 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=10806 freq=72000291 heater_bed: target=110 temp=110.0 pwm=0.627 sysload=0.35 cputime=1718.473 memavail=3110540 print_time=753.776 buffer_time=2.583 print_stall=0 extruder: target=260 temp=259.8 pwm=0.591
Lost communication with MCU 'mcu'
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

Flo-Kra, 2022-02-11:
I have exactly the same issue since I updated Klipper and switched from Octoprint to Moonraker/Mainsail some weeks ago. 

Some prints work fine, and then occasionally i get a Klipper shutdown error - log file see below. 
Sometimes it happened after some minutes, sometimes after many hours of printing. 

```

Got EOF when reading from device
...
Timeout with MCU 'mcu' (eventtime=22619.089863)
Transition to shutdown state: Lost communication with MCU 'mcu'
```

For me it looks like the MCU firmware hangs up or resets for some reason. 
I also had to restart the entire system in some (not all) cases as the USB serial port was gone, so no firmware restart possible any more. 
I tried different good quality USB cables, improved RPi cooling, but all with no success. 
I don´t suspect it is caused by an USB port reset from RPi side as the webcam still works when the issue occurs. 

Used Klipper for 3 months with Octo and had no such problems ever. Not a single unexpected lost print in that time. I don´t know exactly which build i had, but since I started with Klipper in october 21 it must have been release v0.10.0 or a commit shortly after. 
Before that I have been using the printer with Octoprint and Marlin firmware for a year - without any USB/MCU connectivity issues.

My printer and system: 
- Creality CR-10 V3 with original Creality v2.5.2 "silent board" (8bit ATmega2560, TMC2208)
- Modifications: added BLtouch, connected extruder fan to unused Mosfet output and controller fan via Mosfet board to unused GPIO - configured Klipper accordingly and everything worked as expected
- built Klipper with advanced options -> initial pin states for both of the fans, so that they start running on any MCU restart
- dedicated RPi 4 4GB running Raspberry OS Buster, original Raspberry 4 PSU
- 3.5" SPI touch TFT
- Klipper, Mainsail, Moonraker + KlipperScreen installed using KIAUH
- USB cable with 5V pin isolated as the Creality board unfortunately backpowers the RPi

First time the error occured with v0.10.0-237 (28b568a6). 
Updated Klipper/Klippy to v0.10.0-244 (15ffa859) but got the same issue (from my perception more often since then, but I am not sure). 

I will now try again with v0.10.0 (2a964a46), both MCU and Klippy, hoping that it will then work as before. 
I am relatively sure that it is not caused by Moonraker, nor Mainsail, even given that with Octoprint the approach is very different because it doesn´t use virtual SD card as Moonraker/Mainsail does. Anyhow, the log messages don´t point in that direction. 

klippy.log:

```

Stats 6620.7: gcodein=0 mcu: mcu_awake=0.027 mcu_task_avg=0.000175 mcu_task_stddev=0.000173 bytes_write=740593 bytes_read=436265 bytes_retransmit=9 bytes_invalid=0 send_seq=26956 receive_seq=26955 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=9 stalled_bytes=0 freq=16000214 heater_bed: target=65 temp=65.0 pwm=0.333 sd_pos=80803 sysload=0.52 cputime=91.768 memavail=3490716 print_time=2977.715 buffer_time=3.468 print_stall=0 extruder: target=200 temp=199.8 pwm=0.481
Stats 6621.7: gcodein=0 mcu: mcu_awake=0.027 mcu_task_avg=0.000175 mcu_task_stddev=0.000173 bytes_write=740692 bytes_read=436409 bytes_retransmit=9 bytes_invalid=0 send_seq=26963 receive_seq=26962 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=14 stalled_bytes=0 freq=16000214 heater_bed: target=65 temp=65.0 pwm=0.113 sd_pos=80803 sysload=0.52 cputime=91.795 memavail=3490540 print_time=2977.715 buffer_time=2.467 print_stall=0 extruder: target=200 temp=199.8 pwm=0.483
Stats 6622.7: gcodein=0 mcu: mcu_awake=0.027 mcu_task_avg=0.000175 mcu_task_stddev=0.000173 bytes_write=741416 bytes_read=436618 bytes_retransmit=9 bytes_invalid=0 send_seq=26983 receive_seq=26982 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=9 stalled_bytes=0 freq=16000213 heater_bed: target=65 temp=65.0 pwm=0.386 sd_pos=80869 sysload=0.52 cputime=91.829 memavail=3490604 print_time=2980.567 buffer_time=4.318 print_stall=0 extruder: target=200 temp=199.9 pwm=0.422
Stats 6623.7: gcodein=0 mcu: mcu_awake=0.027 mcu_task_avg=0.000167 mcu_task_stddev=0.000167 bytes_write=741520 bytes_read=436763 bytes_retransmit=9 bytes_invalid=0 send_seq=26990 receive_seq=26989 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=9 stalled_bytes=0 freq=16000214 heater_bed: target=65 temp=65.0 pwm=0.267 sd_pos=80869 sysload=0.47 cputime=91.856 memavail=3490632 print_time=2980.567 buffer_time=3.317 print_stall=0 extruder: target=200 temp=199.9 pwm=0.409
Stats 6624.7: gcodein=0 mcu: mcu_awake=0.027 mcu_task_avg=0.000167 mcu_task_stddev=0.000167 bytes_write=741624 bytes_read=436907 bytes_retransmit=9 bytes_invalid=0 send_seq=26997 receive_seq=26996 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=9 stalled_bytes=0 freq=16000214 heater_bed: target=65 temp=65.0 pwm=0.235 sd_pos=80869 sysload=0.47 cputime=91.886 memavail=3490612 print_time=2980.567 buffer_time=2.315 print_stall=0 extruder: target=200 temp=200.1 pwm=0.345
Stats 6625.7: gcodein=0 mcu: mcu_awake=0.027 mcu_task_avg=0.000167 mcu_task_stddev=0.000167 bytes_write=742308 bytes_read=437106 bytes_retransmit=9 bytes_invalid=0 send_seq=27015 receive_seq=27014 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=9 stalled_bytes=0 freq=16000214 heater_bed: target=65 temp=65.0 pwm=0.380 sd_pos=80935 sysload=0.47 cputime=91.912 memavail=3490640 print_time=2983.442 buffer_time=4.189 print_stall=0 extruder: target=200 temp=200.1 pwm=0.323
[... cut ...] p=200.1 pwm=0.358
Stats 6659.8: gcodein=0 mcu: mcu_awake=0.021 mcu_task_avg=0.000150 mcu_task_stddev=0.000142 bytes_write=743908 bytes_read=437943 bytes_retransmit=9 bytes_invalid=0 send_seq=27070 receive_seq=27070 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=58 stalled_bytes=1223 freq=16000213 heater_bed: target=65 temp=65.0 pwm=0.296  sysload=0.26 cputime=92.724 memavail=3490060 print_time=2992.224 buffer_time=0.000 print_stall=0 extruder: target=200 temp=200.1 pwm=0.358

```

Flo-Kra, 2022-02-11:
> Github is not the intended place for such a support request. Please open a topic on https://klipper.discourse.group and post your respective `klippy.log` files showing the issue.

I can´t speak for the one who initially opened this issue, but my intention was NOT to get support, but as I stated, after analyzing the problem and trying many possible solutions without success, I think it is a bug that leads to stability issues in the AVR firmware. So opening an issue on github (or in this case commenting to an existing one stating the very same problem) was the obvious next step for me. I wrote what I have tried to avoid people thinking this is a support request. :-/

Sineos, 2022-02-11:
The general philosophy taken here is to discuss such topics on Discourse and determine if it is a bug, a configuration issue or some hardware issue.
In any case a full and unmodified `klippy.log` showing the issue is needed for further analysis.

On the other hand, if you have already analyzed the behavior and found a bug in the code, then Github is the right place to point it out and to go into the discussion of a potential solution strategy.

Flo-Kra, 2022-02-11:
OK I see. 
I couldn´t track the issue down to an actual bug, so you´re right, I should have asked in the support forum first, but I thought it´s possibly a known bug as someone else had already opened an issue here describing the exact same behaviour. 

Lets see - debugging someone else´s code you´ve never worked on is hard, especially on Arduino and when you have an intermittent error that can´t really be reproduced, as you don´t really know what exactly is triggering it... Unfortunately there is really not much in the log that gives me a clue on where to look precisely. 
As it worked well for me before updating I am now testing with v0.10.0 release tagged commit and unchanged configuration to see if it works as it did before. Will start a discussion on Discourse as you told me as soon as I know more, if needed. 

Thank you for your support so far.

closed 2022-02-13 by github-actions[bot]