Klipper3d/klipper #5218: Pi won't communicate given USB-ID

bogeyf14, 2022-02-01:
##Issues: 

Could not get klipper via mainsail to communicate with BTT Octopus
[image]

##Conditions:
Used the following command to acquire the USB-id
[image]

and entered it into the config
[image]

Klipper would not connect to the board. 

##Solution 

Once I changed the "serial:" line to 
"serial: dev/ttyACM0"

The boards would then communicate

jakep82, 2022-02-01:
Remove {} from your serial device line.

bogeyf14, 2022-02-14:
🤦

closed 2022-02-14 by bogeyf14