Klipper3d/klipper #5226: Klipper unnecessarily slow on curved segments

hoffbaked, 2022-02-02:
On curved paths made of many small segments, such as gyroid infill, the junction deviation calculation is governing the velocity, when the move_centripetal_v2 approximation is a more appropriate speed limit when the segments are very small.

I tried adding a hard cutoff where move_centripetal_v2 is used below 1mm, and R * self.accel is used otherwise, and the velocity reflected the acceleration setting much better. I'm not sure where the cutoff should be, and if it makes sense to blend between the two in some range of segment sizes.

hoffbaked, 2022-02-02:
Here's a quick and dirty implementation the also takes the angle into account:
`        # Approximated circle must contact moves no further away than mid-move
        tan_theta_d2 = sin_theta_d2 / math.sqrt(0.5*(1.0+junction_cos_theta))       
        # Use circle approximation for short segments with angles > 135 degrees
        if self.move_d < 1.0 and junction_cos_theta < -0.7071067812:
            move_centripetal_v2 = .5 * self.move_d * tan_theta_d2 * self.accel
        else:
            move_centripetal_v2 = R * self.accel
        if prev_move.move_d < 1.0 and junction_cos_theta < -0.7071067812:
            prev_move_centripetal_v2 = .5 * prev_move.move_d * tan_theta_d2 * prev_move.accel
        else:
            prev_move_centripetal_v2 = R * prev_move.accel
         # Apply limits
        self.max_start_v2 = min(
            move_centripetal_v2, prev_move_centripetal_v2,
            extruder_v2, self.max_cruise_v2, prev_move.max_cruise_v2,
            prev_move.max_start_v2 + prev_move.delta_v2)
        self.max_smoothed_v2 = min(
            self.max_start_v2
            , prev_move.max_smoothed_v2 + prev_move.smooth_delta_v2)
`

With things like gyroid infill, it may be the case the self.accel/2 is a more reasonable limit.

jakep82, 2022-02-02:
This looks like a dup of https://github.com/Klipper3d/klipper/issues/4228.

hoffbaked, 2022-02-02:
Similar, but mine is specifically about very small segment sizes and smooth curves, whereas the other is talking about much larger segments with courser geometry.

—David

Sineos, 2022-02-18:
Discussions or questions about the code (if you intend to develop or improve Klipper) are best placed on Discord https://klipper.discourse.group as the Devs are reading there too.
GH is only used to share development results, so it is very unlikely to get any further responses here.

closed 2022-02-20 by github-actions[bot]