Klipper3d/klipper #5228: Can't extrude over 0.640mm^2, changing settings seem to do nothing

bitfabrikken, 2022-02-03:
I'm trying to print with a 1mm nozzle, very thick lines, but klipper keeps aborting the print with this error: 
Move exceeds maximum extrusion (0.870mm^2 vs 0.640mm^2)

I've set the following parameters to troubleshoot (overkill yes, but why not when troubleshooting):
```
max_extrude_only_distance=150.0
max_extrude_cross_section=16.0
nozzle_diameter=1
```

After a restart and attempt to print again, the error reoccurs, with the same value of 0.640mm^2 in the error.
Shouldn't that 0.640mm^2 value change when I've changed the max_extrude_only_distance parameter?

I fear this might be a bug. Everything I've Googled suggests increasing max_extrude_cross_section - but the value in the error doesn't change accordingly.

Attached klippy.log with code M112 run, error occurs on line 10619: [klippy.log](https://github.com/Klipper3d/klipper/files/7992719/klippy.log)

jakep82, 2022-02-03:
A search of your log finds no configuration for max_extrude_cross_section, and 2 instances of nozzle_diameter with both set at 0.4.  You've either edited the wrong config file, or didn't restart between changes to the file.

bitfabrikken, 2022-02-04:
@jakep82 indeed I was changing the wrong file - Voron2_Octopus_Config-20220120_135927.cfg
Sorry for wasting your time, but thanks for pointing it out :)

closed 2022-02-04 by bitfabrikken