Klipper3d/klipper #5229: BLTouch does not work after a G91

OasisOfChaos, 2022-02-03:
Initially found in the latest MainsailOS but reproduced in Klipper Git clone version: 'v0.10.0-249-g427f4b45'

Problem: after issuing a G90/G91 the BLTouch probe does not respond to commands anymore. After running a G28/Z_TILT_ADJUST the probe works again.

Steps to reproduce (sending them all from the mainsail web interface):
- send ```G28```;
- printer does its homing thing and probe deploys normally;
- send ```BLTOUCH_DEBUG COMMAND=self_test```;
- probe deploys as expected;
- send ```G91``` (and maybe something like ```G1 X-1 F6000``` but that does not seem to matter);
- send ```BLTOUCH_DEBUG COMMAND=self_test```;
- probe does not deploy;
- send ```G28``` and it works again.

When the probe is not reacting, a ```BLTOUCH_DEBUG COMMAND=reset``` does not do anything either.

Logfile: [klippy.log](https://github.com/Klipper3d/klipper/files/7993580/klippy.log)

Sineos, 2022-02-04:
Even running:
```
BLTOUCH_DEBUG COMMAND=self_test
BLTOUCH_DEBUG COMMAND=self_test
```
does not work.

running:
```
BLTOUCH_DEBUG COMMAND=self_test
BLTOUCH_DEBUG COMMAND=pin_down
BLTOUCH_DEBUG COMMAND=self_test
```

again works.

Weird but no practical meaning or impact, IMO

OasisOfChaos, 2022-02-05:
I've had some issues with Z_TILT_ADJUST and the probe that didn't deploy. That's why I came to this. But I cannot reproduce this so that probably had some other cause. So, indeed no impact. Closing the issue and thanking you for replying!

closed 2022-02-05 by OasisOfChaos