Klipper3d/klipper #5231: Klippy-state: error

jensvanoverberghe1, 2022-02-04:
I've trief already for several times to put Klipper firmware onto my Creality Ender 3. Every time I'm done with the steps that need to be done, my mainsail keeps showing this text:

mcu 'mcu': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer

Anyone wo can help me out pls
[moonraker.log](https://github.com/Klipper3d/klipper/files/8003215/moonraker.log)
[klippy (1).log](https://github.com/Klipper3d/klipper/files/8003216/klippy.1.log)
?

theopensourcerer, 2022-02-04:
Please see the page in the docs about Contact.

https://github.com/Klipper3d/klipper/blob/master/docs/Contact.md#help-it-doesnt-work

Your log is full of:

```
error: mcu 'mcu': Serial connection closed
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
```

I would guess that the mcu is not running. 

But as discussed in the Contact us page we don't use Github for troubleshooting - please try another community forum as suggested.

jensvanoverberghe1, 2022-02-04:
thanks man, do you know how to adjust it to so it can work again? Sorry to send it here, had no way out anymore :(

closed 2022-02-20 by github-actions[bot]