Klipper3d/klipper #5236: BiquBX was addded to Klipper but temp not working

nicedevil007, 2022-02-05:
Hey guys, I was happy to see that the BiquBX's mainboard was added to Klipper.
We started today on configuring everything with the example cfg provided by https://github.com/Klipper3d/klipper/commit/c15d38b973a56978535f36c0fe61f5f8de468248

Unfortunately the bed temperature isn't working at all, so maybe there seems to be still an issue with the board? It was working with marlin before.

[image]

adelyser, 2022-02-06:
There is an issue with the ADC for the STM32H7. I think I have a fix and I'm testing it, hopefully I can get a PR up soon for it.

FYI: There isn't an issue with your hardware, this is a confirmed problem with the ADC.

nicedevil007, 2022-02-06:
Could it be the case, that it is just reading 200°C too much?

adelyser, 2022-02-06:
No, it's an issue with the peripheral being reset when it shouldn't be.

nicedevil007, 2022-02-06:
and that leads to 200°C more on the ADC? wtf, but ty for clearing this up, hope you can fix it, otherwise we can only print PLA xD

adelyser, 2022-02-06:
I have a fix submitted in PR #5239, if you want to try that.

closed 2022-02-06 by KevinOConnor

nicedevil007, 2022-02-07:
So the PR was applied by updating Klipper over WebUI (mainsail)

[image]

Unfortunately it doesn't fix the issue, temp of bed still 220°C without heating anything

[image]

adelyser, 2022-02-07:
The fix has been confirmed by several users now. So you updated klipper, did you clean, rebuild, and re-flash the mainboard? That is required for this fix.

ZeroCoolAT, 2022-02-07:
Hi adelyser,

thanks for the fix, tipps and information to re-flash.
Got it now running (after re-flash) with the correct temperature.

Geat job!

nicedevil007, 2022-02-07:
Yeah it did work for me now as well, just thought that there would be a Warning or Error in the webui that is telling me that we have to reflash (was the case for other firmware changes in the past)... thats why I didn't do it on my first attemp.

Thank you very much!

PR #5239 (2022-02-06): BUG: Fix STM32H7 ADC reset
module: STM32 ADC
Don't reset the ADC peripheral if the clock is already enabled.

Fixes #5236
commit 9174c024 (2022-02-06): stm32: Fix ADC on stm32h7 (#5239)

Don't reset the ADC peripheral if the clock is already enabled.
Fixes #5236