Klipper3d/klipper #5240: Keep hotend fan running when verify_heater trigger

Tahx, 2022-02-07:
Right now the beharior is to shutdown Klipper entirely, but the issue is that the hotend fan stops running, and if the hotend was at 245°C and the mount of the hotend is in plastic, it'll slowly but surely melt and soften the mount until inevitably the screws melt through as it did happen to me.

To me this seem like an important step to workaround this issue.

Thanks for this software.

Sineos, 2022-02-07:
When Klipper shuts down the hot-end fan is turned on.
If it does not work for you, please post on https://klipper.discourse.group/ together with your `klippy.log` showing the issue.

Tahx, 2022-02-07:
> When Klipper shuts down the hot-end fan is turned on. If it does not work for you, please post on https://klipper.discourse.group/ together with your `klippy.log` showing the issue.

Will check that once I get the chance to have it happening again and post on Discourse with the requested log.

closed 2022-02-07 by Tahx