Klipper3d/klipper #5241: Homing failed due to printer shutdown

M7medd, 2022-02-07:
hello
first sorry about my english 
i run klipper on raspbery pi 3 and btt octopus v1.1 when i trying homing if there any retract distance in any one of the axes the printer restart immediatly after retract i set retract distance for all axes to zero and homing done when i tried to bed mesh calibrate or screw tilt calibrate the issue show up again ((Probing failed due to printer shutdown))
please i need your help fast
[klippy.log](https://github.com/Klipper3d/klipper/files/8017953/klippy.log)

dallerman, 2022-02-07:
"Transition to shutdown state: MCU 'mcu' shutdown: Timer too close
This often indicates the host computer is overloaded. Check
for other processes consuming excessive CPU time, high swap
usage, disk errors, overheating, unstable voltage, or
similar system problems on the host computer."

On line 1323 of the logfile you can see the following lines. PI3 is not the most powerful machine so try a clean install of your PI config

M7medd, 2022-02-07:
okay i tried pi 4 with new sd card and same issu
[klippy.log](https://github.com/Klipper3d/klipper/files/8018382/klippy.log)
e

dallerman, 2022-02-07:
Did you use the same install as for PI3 or did you do a clean install?

M7medd, 2022-02-07:
i did a clean install with new sd card

M7medd, 2022-02-07:
@dallerman okay i found the problem
at the begining in the make menu of klipper to make the firmware.bin file the default clock reference is 8 MHZ but the octopus is 12 MHZ
[image]
[image]
thanks for your help 
i'll close it

closed 2022-02-07 by M7medd