Klipper3d/klipper #5244: Issue of today's update. Klipper is constantly restarting even in the middle of a printer

pccabral, 2022-02-08:


Gilabite, 2022-02-08:
wow what a retarded way to post something here, talk about lack of reading anything about posting an issue here, documentation or anything. no log, no info about your hardware literally NOTHING how would you think someone could help you with anything? Here let me do the minimal for you https://github.com/Klipper3d/klipper/blob/master/docs/Contact.md

pccabral, 2022-02-08:
Actually is shutting down. Not restarting

Gilabite, 2022-02-08:
wow real helpful there bud, try reading the link, this is not the place for this

pccabral, 2022-02-08:
I'm sorry. I'm new in GithumGithub. No offense intended. I just updated my version for:
Klipper: v0.1-.0256
Moonraker is v0.7.1-356
Klipperscreen V0.2.0-85
Mainsal v2.1.
I use a Octopus from Bigtreetech
Never had a problem like this. I don't know what more info you need. Feel free to ask anything and once again I'm sorry...Old guy new technology.;-)

Gilabite, 2022-02-08:
You are getting Heater extruder not heating at expected rate error either you have a hardware issue like your thermistor is broken or a loose wire on it. Was this running and you just did an upgrade, is it a new build, have you screwed with any of the the connections etc. Your severe lack of info isnt helping at all. Again not really the place for this, not a klipper issue but user error and should be on the discord/discourse to get help like this refer to the link I posted before

pccabral, 2022-02-08:
It was running perfectly. I have it built by 2 month now. Last was yesterday night. Print every time I get a break from work. No mechanical/physical problem

pccabral, 2022-02-08:
It was running perfectly. I have it built by 2 month now. Last was yesterday night. Print every time I get a break from work. No mechanical/physical problem. I watch my first line closely, and I did not found any temperature issue. I showing what define (250C)... When I heat up it ramp up normally. How can I back out the last update and try it again? The older version was working fine, just want to check if is an software issue?

pccabral, 2022-02-08:
Let me rephrase. (Wrote poorly) I end my build 2 month ago and have been printing happily and without problem until yesterday, today when I was about to do my first print of the day I got this problem.

Gilabite, 2022-02-08:
you're still missing that this isnt the place for help like this part

pccabral, 2022-02-08:
Sorry. I won't bother this channel any more... You already identified where the problem is (I can understand the Klipy file, not my forte). I will recheck all connections again and I will go from that. If not I will read the instructions better and try to find the proper channel.

pccabral, 2022-02-08:
Thanks for all your help and patience

closed 2022-02-08 by pccabral

KevinOConnor, 2022-02-09:
@Gilabite - it is important to me that all communications on this forum be positive, upbeat, and courteous.  If you're frustrated with a ticket, then I recommend you "step back" from it.  Please don't post your frustrations in a negative way on this forum.

-Kevin

Gilabite, 2022-02-09:
@KevinOConnor sorry I will refrain from that, I do try to help when I can but do have a lack of patience sometimes, something I need to work on, not just here.