Klipper3d/klipper #5253: Double Z axis

TonyM4119, 2022-02-09:
So I'm having an issue I got klipper setup on my ender 3 with the skr 2 and tmc2209 drivers and everything seemed to be working well but when I did my first test print my z axis moved twice as high so instead of my test cube being 20x20x20 it was 20x20x40 in the z and I can't figure out why

Jonix71, 2022-02-10:
Share your Z config. But it sounds like you have wrong rotation_distance/microsteps/full_steps_per_rotation in the config.  Are you running TMC2209 in UART?

TonyM4119, 2022-02-10:
I am thought it was the wrong rotation distance just not sure what to make it,, I've tried it with out running uart this config has the z axis in uart only I haven't tested printed yet tho. Here's the link to the config

https://drive.google.com/file/d/1EwU53uiOXVZBddEaSQ0wD5pykcb0tf1f/view?usp=drivesdk

bevanweiss, 2022-02-16:
You could just take the rotation_distance for both z and z1 up to 16 (currently at 8).
This will let Klipper know that one rotation of the stepper results in 16mm of movement (and hence it will spin the motors 50% less than it currently does.. moving 50% less distance than it currently does).
This should convert your 40mm Z distances down to 20mm Z distances, matching the cube.

Sineos, 2022-02-18:
rotation_distance is depending on the stepper you are using.
Typically the values in the provided example configuration are 1.8° stepper. If you are using 0.9° stepper, the rotation distance needs to be doubled for the axes that do so. Ref: https://www.klipper3d.org/Rotation_Distance.html

Such configuration issues are typcially handled not on GitHub. Please open a respective support request on https://klipper.discourse.group

closed 2022-02-20 by github-actions[bot]