Klipper3d/klipper #5255: M104 Sxxx Tx - Klipper reports "Extruder not configured" under single extruder slicing

jalanjarosz, 2022-02-11:
This is in reference to changes in: https://github.com/Klipper3d/klipper/pull/5143

Klipper v0.10.0-263-ge3cbe7ea

**Hardware:** 
Single hot end/heater, multiple extruders, using 'extruder_stepper' for additional non-synced steppers (2in/1out non-mixing)

**Slicer:**
SuperSlicer with multiple extruders and Klipper gcode flavor

**Klipper extruder and extruder_stepper setup:**
```
[extruder]
step_pin: PC1
dir_pin: PC3 
enable_pin: !PC7
microsteps: 16
rotation_distance: 22.024
full_steps_per_rotation: 200
gear_ratio: 50:17
nozzle_diameter: 0.400
filament_diameter: 1.750
max_extrude_only_distance: 950.0
max_extrude_only_velocity: 250
max_extrude_only_accel: 2750
heater_pin: PB4
sensor_type: ATC Semitec 104GT-2 
sensor_pin: PK5 
control: pid
pid_Kp: 28.114
pid_Ki: 1.562
pid_Kd: 126.511
min_temp: 170
max_temp: 300

[gcode_macro T0]
gcode:
    # Deactivate stepper in my_extruder_stepper
    SYNC_STEPPER_TO_EXTRUDER STEPPER=belted_extruder EXTRUDER=
    # Activate stepper in extruder
    SYNC_STEPPER_TO_EXTRUDER STEPPER=extruder EXTRUDER=extruder
	
[extruder_stepper belted_extruder]
extruder: extruder
step_pin = PA4
dir_pin = PA6
enable_pin = !PA2
microsteps: 16
rotation_distance: 22.5
full_steps_per_rotation: 200
gear_ratio: 4:1

[gcode_macro T1]
gcode:
    # Deactivate stepper in extruder
    SYNC_STEPPER_TO_EXTRUDER STEPPER=extruder EXTRUDER=
    # Activate stepper in my_extruder_stepper
    SYNC_STEPPER_TO_EXTRUDER STEPPER=belted_extruder EXTRUDER=extruder
```

When SuperSlicer generates gcode for a change in temp at 2nd layer, the gcode like is:
`M104 S200 T1`

Klipper extruder.py, under cmd_M104 will report "Extruder not configured".  
```
    def cmd_M104(self, gcmd, wait=False):
        # Set Extruder Temperature
        temp = gcmd.get_float('S', 0.)
        index = gcmd.get_int('T', None, minval=0)
        if index is not None:
            section = 'extruder'
            if index:
                section = 'extruder%d' % (index,)
            extruder = self.printer.lookup_object(section, None)
            if extruder is None:
                if temp <= 0.:
                    return
                raise gcmd.error("Extruder not configured")
        else:
            extruder = self.printer.lookup_object('toolhead').get_extruder()
        pheaters = self.printer.lookup_object('heaters')
        pheaters.set_temperature(extruder.get_heater(), temp, wait)
```

Printing stops, but heaters remain on and steppers active.

Note: M104 S200 T1 works when using [extruder] and [extruder1] with shared_heater.  I changed Klipper config due to shared_heater deprecation notice.

Current fix is gcode macro:
```
[gcode_macro M104]
description: Replaces built-in gcode to not specify Tx due to single extruder
rename_existing: M104.1
gcode:
    {% set s = params.S|default(0)float %}
    {% set t = params.T|default(0)|int %}
    {% if 'S' in params %}
      {%  if 'T' in params %}
        M104.1 S{s}
      {% else %}
        M104.1 S{s}
      {% endif %}
    {% endif %}

```

jalanjarosz, 2022-02-11:
[klippy.zip](https://github.com/Klipper3d/klipper/files/8050684/klippy.zip)

KevinOConnor, 2022-02-11:
It's not clear to me if you are reporting a regression (new klipper code with same config now results in an error), or if you are reporting that the behavior changed after modifying your config file (by removing `shared_heater`).

It seems you are reporting the latter.  This change in behavior is normal - the `M104 S200 T1` command is no longer valid as there is only one extruder heater.  You can either change the slicer to not emit heating commands for two extruders, or create an `M104` macro override to fixup the slicer commands.  You can also continue to use the `shared_heater` setting - at least until it is removed from Klipper.

Probably best to move further discussions to Klipper Discourse - https://www.klipper3d.org/Contact.html

-Kevin

jalanjarosz, 2022-02-11:
@KevinOConnor,

I wasn't sure if this was a break in code, or intended functionally.  I'll be a little clearer in the future. 
Thanks for responding so quickly.  I'll move this over to Discourse after I perform a bit more testing and check out the slicer for heating options.

Thanks
-James

closed 2022-02-11 by jalanjarosz

Roycinger, 2022-06-23:
> @KevinOConnor,
> 
> I wasn't sure if this was a break in code, or intended functionally. I'll be a little clearer in the future. Thanks for responding so quickly. I'll move this over to Discourse after I perform a bit more testing and check out the slicer for heating options.
> 
> Thanks -James

Hi James,
did you manage to figure out a solution without using a custom M104 macro or how tell SuperSlicer not to emit heating commands for two extruders as @KevinOConnor said? If yes, how would you use filaments with 2 different temperature profiles then?
To me, the removal of functionability of the format `M104 S200 T1` is pretty odd. As there's no direct "replacement" of interpretation of the slicers command, the removal of this functionality leads to a hole in Klipper imho. Usually tinkering / fiddling around like this is not the structural way of Klipper that I'm used to.