Klipper3d/klipper #5265: undocumented config parameters

cooked, 2022-02-15:
I can't locate any of the buffer_time_xxx config parameter in the Configuration reference nor in the config examples 
[image]

Piezoid, 2022-02-16:
These parameters are not intended for end users. I think they are kept as configurable parameters for the rare cases of debugging and workarounds.

However, I agree it would be a good idea to document them.
Some time ago I've started writing developer documentation about the timings and flushing logic. I didn't went though because of greys area in my understanding. When I find time to clean up my draft, I'll do a "help wanted" PR.

cooked, 2022-02-16:
I @Piezoid thanks for the repIy, I thought that might be the case. I found myself indeed going down into these klipper core topics, that might be daunting for a newcomer developer. My "dev notes" have grown a lot and I'd like for them to be more structured as well. This is sidetracking a bit but let me know if you would be interested in working out your notes and mine together... we could probably use each others to improve our understanding (I for sure also have grey areas for which I could use a 2nd pair of eyes ) and come up in the end with good additions to expand the official docs.

cooked, 2022-02-18:
moved to the forum https://klipper.discourse.group/t/undocumented-buffer-time-parameters/2074

closed 2022-02-18 by cooked