Klipper3d/klipper #5266: New klipper install Unable to parse option 'max_z_accel' in section 'printer'

technovikingurgh, 2022-02-16:
Just finished my first install of Klipper on my ender 3 pro with 4.2.7 board. Now i am running in to issues with configuring my build. I am very new to coding and editing things like this. I saw another post of someone having the same issue and it was solved by removing the line. I just don't understand how to do that myself.  I have attached my log if someone would be able to point out the problems with my build that would be greatly appreciated. 
Cheers! 

[klippy.log](https://github.com/Klipper3d/klipper/files/8076857/klippy.log)

bevanweiss, 2022-02-16:
I think you have something 'weird' with your config that just happens to be around the 'max_z_accel' line.
So.  Delete both the 'max_z_accel' and the '[gcode_macro PAUSE]' lines completely.
And then TYPE them in (DO NOT COPY/PASTE).  You should leave a nice blank line between them also.

I believe your copy-paste of the gcode macro has brought in some unicode character or other hidden character making the carriage return on the max_z_accel 'borked'.

```
[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 3000
max_z_velocity = 5
max_z_accel = 100
	[gcode_macro PAUSE]
description = Cancel the actual running print
rename_existing = CANCEL_PRINT_BASE
```

Sineos, 2022-02-22:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support request on https://klipper.discourse.group/ if the issue still persists. There you will also find multiple threads already with such issue and how to solve them.

closed 2022-02-24 by github-actions[bot]