Klipper3d/klipper #5268: Possible to home Z on full steps with all Z steppers after z_tilt?

StefanMeGit, 2022-02-16:
I just tried to set my Z axis to home on a full step after z tilt, but i count make it work. Is this function provided in klipper? Would be awesome... 
```
[endstop_phase stepper_z]
#endstop_accuracy:
#   Sets the expected accuracy (in mm) of the endstop. This represents
#   the maximum error distance the endstop may trigger (eg, if an
#   endstop may occasionally trigger 100um early or up to 100um late
#   then set this to 0.200 for 200um). The default is
#   4*rotation_distance/full_steps_per_rotation.
#trigger_phase:
#   This specifies the phase of the stepper motor driver to expect
#   when hitting the endstop. It is composed of two numbers separated
#   by a forward slash character - the phase and the total number of
#   phases (eg, "7/64"). Only set this value if one is sure the
#   stepper motor driver is reset every time the mcu is reset. If this
#   is not set, then the stepper phase will be detected on the first
#   home and that phase will be used on all subsequent homes.
#endstop_align_zero: False
#   If true then the position_endstop of the axis will effectively be
#   modified so that the zero position for the axis occurs at a full
#   step on the stepper motor. (If used on the Z axis and the print
#   layer height is a multiple of a full step distance then every
#   layer will occur on a full step.) The default is False.
```
I guess after i make the command for z_tilt its not a full step anymore?
[klippy (7).log](https://github.com/Klipper3d/klipper/files/8078552/klippy.7.log)

bevanweiss, 2022-02-16:
What did you try exactly?  The config you've provided doesn't show that you've tried anything.
I would have expected you to have enabled 'endstop_align_zero: True', and then provided test results which indicate that it doesn't work for you.

Perhaps you might have even tried 'trigger_phase: 0/16' to see if that worked.

And I also assume that you read up from here https://www.klipper3d.org/Endstop_Phase.html
and that you ran the ENDSTOP_PHASE_CALIBRATE command and have some results to show for this.

Sineos, 2022-02-16:
See an existing discussion here: https://klipper.discourse.group/t/rounding-to-nearest-full-step-when-homing/1216/6

As already directed when you opened this issue: Please be so kind to post support or feature requests on discourse. Many thanks

closed 2022-02-18 by github-actions[bot]