Klipper3d/klipper #5269: Klipper Shut Down during Print at Speed >70

sohejl, 2022-02-16:
Hi,
hopefully someone has an idear whats wrong. This issue appears, as of now only at print speeds above 70mm/s.

Systemcomponents:
- Anet A8 BVL upgrade
- BTT SKR Pico Mainboard
- Mainsail and Klipper on an Raspberry Pi 4 2GB, software fully updated
- Connection Between Raspberry and SKR Pico via USB
- Raspberry Cam 
- Slicer SuperSlicer
- printer.cfg (renamed to txt), moonraker and klipper log attached
[klippy (1).log](https://github.com/Klipper3d/klipper/files/8078519/klippy.1.log)
[moonraker (1).log](https://github.com/Klipper3d/klipper/files/8078521/moonraker.1.log)
[printer (1).txt](https://github.com/Klipper3d/klipper/files/8078532/printer.1.txt)

Issue:
As of now I haven't had the problem with speeds up to 70mm/s. With speeds higher than that, last time with the pressure advance test, as you will find in the log files (100mm/s) the printing stops somewhere during print. In Mainsail I get the info "Klipper Shut Down".
I whent through the log files, but didn't find anything which would help me to indentify the source of the problem.
At the beginning I had the impression that the x an y stepper motors get very hot and it might be an overheat protection or so, there fore I reduced the stepper current from 0.8 to 0.65. The stepper motors seem not running so hot anymore but otherwise there was no change regarding the Klipper Shut down at higher speeds.

Thanks in advance for the Help.

jakep82, 2022-02-16:
From your log :

`Heater extruder not heating at expected rate
Transition to shutdown state: Heater extruder not heating at expected rate`

`See the 'verify_heater' section in docs/Config_Reference.md
for the parameters that control this check.`

This has nothing to do with speed.  Your part fan is likely cooling the hotend and it can't get back to temperature fast enough.  If you need help in the future, please post on the Discord or Discourse.  We don't use Github for help requests.

<https://github.com/Klipper3d/klipper/blob/master/docs/Contact.md#help-it-doesnt-work>

sohejl, 2022-02-16:
Hi and thanks, if you don't mind me asking one last question... Is it normal behaviour  shutting down klipper, instead of turn off the extruder stop print and show a warning?

jakep82, 2022-02-16:
Any heater malfunction is a critical error and Klipper will shut down.

sohejl, 2022-02-16:
Thanks for the help. I will check that out!

closed 2022-02-16 by sohejl