Klipper3d/klipper #5279: Move out of range

Steven2k6, 2022-02-17:
Hey, I set up klipper and I had to change the rotaion distance right.
After I did that, I get this output: "Move out of range: 280.993 267.732 25.300 [15.079]"
This happens after the homing and I tested a bit and I guess it is there, where the printer really starts printing.

ld3300, 2022-02-17:
Hi @Steven2k6,

This is almost always either a configuration problem or an issue with relative vs absolute position settings.  If you go to to the discord server and post the issue and your printer.cfg in the print-help channel you are likely to get help.  Feel free to also send me the file and I can look it over.  The discord is: [https://discord.com/channels/431557959978450984/802921046452731944](https://discord.com/channels/431557959978450984/802921046452731944)

Steven2k6, 2022-02-17:
Okay, thanks, should I send it to you in this issue?

jakep82, 2022-02-17:
We don't use GitHub for help requests.  Please post this request on Discord or Discourse.

<https://www.klipper3d.org/Contact.html#help-it-doesnt-work>

Steven2k6, 2022-02-17:
Thanks guys, I got help, have a nice day

closed 2022-02-17 by Steven2k6