Klipper3d/klipper #5288: MCU 'mcu' shutdown: Timer too close

AndehX, 2022-02-20:
I have  Tronxy X5SA-Pro with the Chitu v6 board.  I can print for quite a while (like around 12 hours) before randomly getting this error, which ruins my print.  I've attached part of klippy.log.  Can anyone tell me what the hells going on?  It's driving me insane.
[klippy.log](https://github.com/Klipper3d/klipper/files/8103780/klippy.log)

cbc02009, 2022-02-20:
This is best discussed on the discourse forums, not on github. Github is for developer issues.

myung, 2022-02-21:
I have a X5SA and I have same problem. Since few day ago.

AndehX, 2022-02-21:
I've wired up the serial connection from my raspberry pi directly to the mainboard to bypass the USB to serial chip, to see if this helps, as I've read that these cheap boards can use knock-off clone chips.  I'll update if i still have issues.

closed 2022-02-23 by github-actions[bot]