Klipper3d/klipper #5297: Printer resets while printing

sgoum, 2022-02-22:
Hello. 
I have build a printer using Klipper. 
At random times while printing it shuts down.

I get the following message:

MCU 'mcu' shutdown: Timer too close
This often indicates the host computer is overloaded. Check
for other processes consuming excessive CPU time, high swap
usage, disk errors, overheating, unstable voltage, or
similar system problems on the host computer.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

I have checked the unstable voltage by using a powerbank as PS for RPi.
I have nothing else installed besides Mainsail Klipper. 

Any idea what is wrong?

Thank you in advance

sgoum, 2022-02-22:
I thing I found the problem. I have replaced the UART cables until now no problem.

Sineos, 2022-02-22:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support request on https://klipper.discourse.group/ if further assistance is requested

closed 2022-02-24 by github-actions[bot]