Klipper3d/klipper #5299: LPC1769 non working UART3 (P4.29, P4.28)

kpochwala, 2022-02-22:
Serial communication on UART3, pins 4.29 and 4.28 is not working. Switching over to UART0 (P0.3, P0.2) works properly. I've checked electrical connections and they are good - sending some data via this UART using simple mbed test firmware works properly.

Some hints that I've read from documentation, however I'm not too shabby with low-level MCU programming. [LPC17xx User Manual UM10360 p. 298], UART0 is enabled by default, and UART2/3 is not. - that might be some hint. Also enabling peripheral clock is done in different register - PCLKSEL0 for UART0, PCLKSEL1 for UART2/3. 

Also weird thing I found is that this function (`src/lpc176x/main.c`): 
```c
...
// Enable a peripheral clock
void
enable_pclock(uint32_t pclk)
{
    LPC_SC->PCONP |= 1<<pclk;
}
...
```
does not in fact enable peripheral clock but enables power to the peripheral. Clock should be enabled in `PCLKSELx`. 

[klippy.log](https://github.com/Klipper3d/klipper/files/8118571/klippy.log)

KevinOConnor, 2022-02-23:
I'm not sure why it doesn't work.  I don't see anything wrong in the code.

> Clock should be enabled in PCLKSELx.

Due to an errata on the lpc176x, we can't change the clocks at runtime - so we program all clocks to full speed at startup (commits ae89a659 and 45cd3543).  So, the only thing we need to do in `enable_pclock()` is enable the peripheral.

> Is bank_pos really calculated properly?

Looks correct to me.  For P4.29, `bank_pos` will be 9 and `pin_pos` will be 26 - thus `pinsel[bank_pos]` will be 0x4002C024 (0x4002C000 + 9 * 4) and `pinmode[bank_pos]` will be 0x4002C064 (0x4002C040 + 9 * 4) - which are both correct.

-Kevin

KevinOConnor, 2022-02-23:
If you make this change, does it improve things?  (The current code looks quite strange.)

```diff
--- a/src/lpc176x/serial.c
+++ b/src/lpc176x/serial.c
@@ -75,8 +75,8 @@ void
 serial_init(void)
 {
     // Setup baud
-    LPC_UARTx->LCR = (1<<7); // set DLAB bit
     enable_pclock(PCLK_UARTx);
+    LPC_UARTx->LCR = (1<<7); // set DLAB bit
     uint32_t pclk = get_pclock_frequency(PCLK_UARTx);
     uint32_t div = pclk / (CONFIG_SERIAL_BAUD * 16);
     LPC_UARTx->DLL = div & 0xff;
```

-Kevin

KevinOConnor, 2022-03-01:
Thanks.  I committed this change (commit 7ce409d7).

-Kevin

closed 2022-03-02 by kpochwala