Klipper3d/klipper #5300: GCODE Variables don't actually update until a second calling of macro

adawalli, 2022-02-22:
I believe I have uncovered an issue in klipper, where the set variable command doesn't "take" until a second pass of the macro is executed.

```toml
[gcode_macro CHECK_RAND]
variable_bucket_pos: 4
gcode:
    SET_GCODE_VARIABLE MACRO=CHECK_RAND VARIABLE=bucket_pos VALUE={(range(2) | random)}
    { action_respond_info("location is %s" % (bucket_pos)) }
```
The above code will demonstrate the issue. An "invalid" value of 4 is the initial value for `bucket_pos`. If you look at the set logic, we actually expect a value of either `0` or `1`. However, after every `firmware_restart`, you will get a 4!

[klippy.log.zip](https://github.com/Klipper3d/klipper/files/8119184/klippy.log.zip)

Arksine, 2022-02-22:
This is expected behavior, as it is how templates work.  The template is evaluated before gcode commands are run, so `action_respond_info()` is called before SET_GCODE_VARIABLE.

adawalli, 2022-02-22:
I understand in the case of printing, but in the case of using this it doesn't work here though. This was the code where I first noticed the problem: https://github.com/VoronDesign/VoronUsers/blob/master/printer_mods/edwardyeeks/Decontaminator_Purge_Bucket_%26_Nozzle_Scrubber/Macros/nozzle_scrub.cfg#L174

Arksine, 2022-02-22:
It is not possible to use `SET_GCODE_VARIABLE` to change `bucket_pos` and have the change reflected during evaluation of the template.

All statements within `{ }` or `{% %}` are run during evaluation, however the rendered result (ie: gcode commands such as SET_GCODE_VARIABLE) are not run until after evaluation.

adawalli, 2022-02-22:
Thanks for the insight!!

closed 2022-02-22 by adawalli

themacboy, 2022-08-01:
Good morning,

one more question about change variables on the fly.

SAVE_VARIABLE to disk works the same way?