Klipper3d/klipper #5302: [Request] Add a lump detector feature to the Hall filament width sensor

rongith, 2022-02-23:
I decided to add the Hall filament width sensor to my printer. It is a great idea.
But my first concern was to try to catch lumps on the filament before it enter the extruder. Some low viscosity filaments have not so good diameters. It is not rare that there are lumps on the filament as big as 2.2mm. Those don't enter hotends (E3Dv6 is 2.0 mm inner diameter and Dragon is 1.9mm ID). If it occurs, you got a 1.5kg part ruined. I would like a feature to pause printer before cutting the lump and restart.
There is a runout filament sensor feature implement triggered by small diameter (1.0 default). But there is no trigger implemented for a ceiling diameter (per exemple 2.0).
Then I done a quite hugly patch for that. I use the same procedure as for the runout:
````
33a34
>         self.lump_dia=config.getfloat('lump_diameter', 2.0)
126c127
<         # Check runout
---
>         # Check anomaly (runout or lump)
128c129
<             self.diameter > self.runout_dia)
---
>             self.diameter > self.runout_dia and self.diameter < self.lump_dia)
```

I'm not fluent at all with git then I didn't open a pull request.

OSHW-Rico, 2022-03-08:
Do you have a picture of that lump on your filament?

rongith, 2022-03-09:
There it is.
[image]

dewi-ny-je, 2022-03-10:
Independently from how often such a case may happen in real life (which is likely not often), I think the feature makes sense: it's consistent with the presence of a lower limit.

rongith, 2022-03-10:
When you have the filament stuck in your hotend, you first think about heatcreep. And it may be not heat creep but a lump. Here I can tell what it is.

Sineos, 2022-03-12:
Please be so kind to either:
* Issue a pull request so that the devs can review and decide
* Open a topic https://klipper.discourse.group/ to further discuss about this
Otherwise this topic will dangle here forever and be eventually closed.

closed 2022-03-12 by rongith