Klipper3d/klipper #5306: Klipper's seemingly absent error handling

SimonBL, 2022-02-23:
I'm not sure if this is a feature or a bug but any error causes controller to disconnect from Octoprint and requires re-connection and some reset. To me, this looks like an issue. Sorry if this is not the best place for this but I don't use Discord and I searched alot to find a solution to this.

Examples: try to move before homing causes a disconnect, request a movement out of bounds causes disconnection. Attempting to feed filament in a cold extruder causes disconnection. Could these minor errors by dealt with by an error message and inaction, rather than disconnect and require a reset.

It's like Klipper has only 1 way to deal with any error, minor or not: disconnect, go into limbo and require reset. Either I'm doing something wrong or error handling by Klipper is absent. 

If this is a feature, I understand it's 'bullet-proof' for safety but it's been a huge source of wasted time while learning to use Klipper and this level of overkill just seems to make no sense. Thanks.

BlackStump, 2022-02-24:
You are barking up the wrong tree bud - It is Octoprint that is disconnecting you!
I suggest reading the Klipper Install docs about setting up Octoprint with Klipper.

jakep82, 2022-02-24:
https://www.klipper3d.org/Installation.html#configuring-octoprint-to-use-klipper

`In the Settings tab, navigate to the "Behavior" sub-tab and select the "Cancel any ongoing prints but stay connected to the printer" option. Click "Save".`

SimonBL, 2022-02-24:
ouff thanks and sorry. There is so much to learn and overwhelming amounts of information to sift through. Now I'm slightly less noob.

closed 2022-02-24 by SimonBL