Klipper3d/klipper #5308: PID_CALIBRATE issue

Akujiproxy, 2022-02-24:
Hi, 

I use klipper on an Ender 3 V2 with a stock hotend. Now I can do a [PID_CALIBRATE HEATER=extruder TARGET=200] without issues but when i try [PID_CALIBRATE HEATER=extruder TARGET=240] so i can dial it in for PETG, I'm getting a [verify_heater] error and the firmware needs to be restarted.. Anybody know whats up with this?

SteveGotthardt, 2022-02-24:
Printer.cfg. What is max temp in extruder section?

Akujiproxy, 2022-02-24:
> Printer.cfg. What is max temp in extruder section?

250

I've found some threads that talk about adding [verify_heater extruder] max_error: 250, which seems to help with regards to PID tuning but I'm not sure what it does exactly and thus not sure if its smart to use this setting

Sineos, 2022-02-24:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support request on: https://klipper.discourse.group/

Akujiproxy, 2022-02-24:
> As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support request on: https://klipper.discourse.group/

Sorry this is the first time i see it. I will close the ticket and go to the discourse instead

closed 2022-02-24 by Akujiproxy