Klipper3d/klipper #5309: controlling a heater with two different thermistors

ahmadmadary, 2022-02-25:
Hi
I am adding a chamber heater to my printer. For this purpose, I have a heating element and two thermistors. Thermistor 1 (T1) measures the temperature of the heater and thermistor two (T2) measures the temperature of the chamber. With thermistor one (T1), I can control the heater temperature to stay constant at the desired temperature using [heater_generic] section. But I want to turn off the heater when chamber temperature (T2) reaches the desired value (something similar to [temperature_fan]).
My question is that: is there a way to have a section for heater that turns on if T2 falls below certain temperature, then using PID control, bring the heater to certain T1 temperature and then shuts down when T2 reaches the desired value? Basically combining having a section such as [temperature_heater] that somehow combine [heater_generic] and the functionality of [temperature_fan]?

Thanks

Sineos, 2022-02-25:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support request on: https://klipper.discourse.group/

ahmadmadary, 2022-02-25:
I apologize fir this.

closed 2022-02-25 by ahmadmadary