Klipper3d/klipper #5328: TMC 2209 run current can only be set to certain values

Itox001, 2022-03-06:
Hello! Found an interesting one when trying to tune the parameters of the driver because my new E3D 0.9 degree motors have some nasty resonance bands that sound like grinding at some speeds. I found that certain run currents make it better or worse, for example from 0.9 A to 1.03 A is a big difference for the better, so I was fine tuning it.

I then found that when trying to set the current on my TMC 2209 drivers (connected via UART on SKR E3 Turbo), Klipper would always select certain values, regardless of the requested value. For example:
[image]

Is this a bug, or there is some limitation? I dug through the documentation and did not find anything related to this.

Thanks and awesome job!
[klippy (2).log](https://github.com/Klipper3d/klipper/files/8193379/klippy.2.log)

Sineos, 2022-03-06:
And as already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/

Itox001, 2022-03-06:
@Sineos oops. Sorry. Closing.

closed 2022-03-06 by Itox001