Klipper3d/klipper #5335: Error loading kinematics 'delta'

PtitDavid, 2022-03-08:
Hello everyone, having problem with my flsun sr, trying to put fluidd clipper on it but it gone wrong…
I tried same setup with octoklipper and mainsail no problem but I want to use fluidd due to many things 
any help would be awesome ;) thanks

Error loading kinematics 'delta'
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/toolhead.py", line 256, in __init__
    self.kin = mod.load_kinematics(self, config)
  File "/home/pi/klipper/klippy/kinematics/delta.py", line 235, in load_kinematics
    def load_kinematics(toolhead, config):
  File "/home/pi/klipper/klippy/kinematics/delta.py", line 17, in __init__
    rail_a = stepper.LookupMultiRail(
TypeError: LookupMultiRail() got an unexpected keyword argument 'need_position_minmax'
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 157, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 143, in _read_config
    m.add_printer_objects(config)
  File "/home/pi/klipper/klippy/toolhead.py", line 599, in add_printer_objects
    config.get_printer().add_object('toolhead', ToolHead(config))
  File "/home/pi/klipper/klippy/toolhead.py", line 264, in __init__
    raise config.error(msg)
Error: Error loading kinematics 'delta'

Sineos, 2022-03-08:
And as already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/

PtitDavid, 2022-03-08:
thank you I delete 👍🏻

closed 2022-03-08 by PtitDavid