Klipper3d/klipper #5343: Done! Here's an example config file: [generic-th3d-ezboard-lite-v1.2.zip](https://github.com/KevinOConnor/klipper/files/6603719/generic-th3d-ezboard-lite-v1.2.zip)

airbornetrooper82573, 2022-03-09:
Done! Here's an example config file: [generic-th3d-ezboard-lite-v1.2.zip](https://github.com/KevinOConnor/klipper/files/6603719/generic-th3d-ezboard-lite-v1.2.zip)
I've included most of the pins from the [pinout map](https://github.com/th3dstudio/UnifiedFirmware/blob/2.0.x/Firmware/Marlin/src/pins/lpc1769/pins_TH3D_EZBOARD.h), so they should be working. HOWEVER, there are some pins in this config file I need to mention:
- EZABL compatibility -  it will probably work with BLTouch pins
- Filament sensor - I don't have nor use one, so I'm not sure if the pins are configured correctly 
- Expansion Header - this is apparently for the "E3D PT100 Temperature Sensor w/ Amplifier Board". I don't have nor use one, so I'm not sure if that will work.
- On-board SD card reader - I doubt anyone will ever use the SD card reader now

And some other notes:
- Most of the printer settings is based off the Ender 3 Pro config, so adjust accordingly with whatever printer you have.
- I used the SKR Mini E3 v2.0 config as a layout template, but I assure you, you won't find those pins here.  
- The run_current for the motors are from TH3D's [configuration_adv.h]
I'm looking for help on how you complied the firmware and if you were able to do it from the Pi or you did it via sd card. 

(https://github.com/th3dstudio/UnifiedFirmware/blob/2.0.x/Firmware/Marlin/Configuration_adv.h). Make sure to enter them correctly in your printer.cfg (600 in Marlin config = 0.600 in Klipper) 
- I am not responsible for potentially destroying your board.

Finally, [here's a test print](https://youtu.be/ZImplD6fM_A)! So far so good!

_Originally posted by @TwoPuncakes in https://github.com/Klipper3d/klipper/issues/2764#issuecomment-855345131_

TwoPuncakes, 2022-03-09:
Basically follow this: https://youtu.be/gfZ9Lbyh8qU?t=247
But make sure you change the settings as outlined in the beginning of generic EZBoard config: https://github.com/Klipper3d/klipper/blob/master/config/generic-th3d-ezboard-lite-v1.2.cfg
```# This file contains common pin mappings for the TH3D EZBoard Lite v1.2.
# To use this config, the firmware should be compiled for the
# LPC1769 with USB communication.

# The "make flash" command does not work on this board. Instead,
# after running "make", copy the generated "out/klipper.bin" file to a
# file named "firmware.bin" on an SD card and then restart the board
# with that SD card.```

airbornetrooper82573, 2022-03-09:
I have Micro-controller Architecture (LPC176x (Smoothieboard) to Processor model (lpc1769 (120 MHz)), Target board used Smoothieware boot loader selected, and Communication interface set to USB. I've copied the file to an SD card with the name firmware.bin. Every time I start the printer, it just brings up a white box on the screen. It never shows anything even after sitting for 5 minutes. I can re-install the marlin unified firmware just fine. But no matter what, it just shows a white screen on my Ender 3 Pro v1 with the EZboard lite v1.2 installed.

TwoPuncakes, 2022-03-09:
I'm not sure what you're doing wrong then. I'd suggest joining the Klipper Discord server and asking help there: https://www.klipper3d.org/Contact.html#discord-chat

airbornetrooper82573, 2022-03-10:
I've tried it with Fluidd running on my Pi as well as Octopi. Neither are working. Would you be able to share your firmware file that I can try?

TwoPuncakes, 2022-03-10:
Just so you know, I'm not responsible for potentially damaging your board: [firmware.zip](https://github.com/Klipper3d/klipper/files/8219966/firmware.zip)

airbornetrooper82573, 2022-03-10:
Thank you, I think I got it connected. The screen on my ender 3 is still showing a white square but I am connected with OctoPi now and will continue the Klipper instructions.

TwoPuncakes, 2022-03-10:
IIRC, the white screen is normal until you configure it yourself in printer.cfg

airbornetrooper82573, 2022-03-11:
> IIRC, the white screen is normal until you configure it yourself in printer.cfg

I can't get my EZABL and/or z stepper to work correctly

```
[stepper_z]
step_pin: P2.2
dir_pin: P1.15
enable_pin: !P1.14
microsteps: 16
rotation_distance: 8
endstop_pin:probe:z_virtual_endstop
position_endstop: 0
position_max: 250
position_min: -1

[probe]
sensor_pin: P1.26
control_pin: P2.4
x_offset: -35
y_offset: -3
z_offset: 0
```

TwoPuncakes, 2022-03-11:
You're out of luck. I don't have an EZABL, so I'm not sure how to fix it. 
Again, I'd suggest joining the Klipper Discord server for help: https://discord.klipper3d.org/
Hell, the TH3D server has their own Klipper channel, ask there: https://discord.th3dstudio.com/

closed 2022-03-14 by github-actions[bot]