Klipper3d/klipper #5345: How can I add a new file.c for my printer?

Marck29, 2022-03-11:
Hi, i'm beginning with klipper and i want to do a new kinematic, i create my file.py ("polear.py") and the file.c (kin_polear.c) and i add it in the _init_.py like this:

```
SOURCE_FILES = [
    'pyhelper.c', 'serialqueue.c', 'stepcompress.c', 'itersolve.c', 'trapq.c',
    'pollreactor.c', 'msgblock.c', 'trdispatch.c',
    'kin_cartesian.c', 'kin_corexy.c', 'kin_corexz.c', 'kin_delta.c',
    'kin_polar.c', 'kin_rotary_delta.c', 'kin_winch.c', 'kin_extruder.c',
    'kin_shaper.c', kin_polear.c, 
]
defs_kin_polear = """
    struct stepper_kinematics *polear_stepper_alloc(char axis);
"""
defs_all = [
    defs_pyhelper, defs_serialqueue, defs_std, defs_stepcompress,
    defs_itersolve, defs_trapq, defs_trdispatch,
    defs_kin_cartesian, defs_kin_corexyprimo, defs_kin_corexz, defs_kin_delta,
    defs_kin_polar, defs_kin_rotary_delta, defs_kin_winch, defs_kin_extruder,
    defs_kin_shaper, defs_kin_polear,
]
```

And I code kin_polear.c like this:

```
#include <stdlib.h> // malloc
#include <string.h> // memset
#include "compiler.h" // __visible
#include "itersolve.h" // struct stepper_kinematics
#include "trapq.h" // move_get_coord

static double xp = 1;
static double xn = -1;
static double yp = 0;
static double yn = 0;
static double
polear_stepper_rn_calc_position(struct stepper_kinematics *sk, struct move *m
                             , double move_time)
{
    struct coord c = move_get_coord(m, move_time);
    static double rn = sqrt((c.x-xn)^2+(c.y-yn)^2);
    return rn;
}

static double
polear_stepper_rp_calc_position(struct stepper_kinematics *sk, struct move *m
                             , double move_time)
{
    struct coord c = move_get_coord(m, move_time);
    static double rp = sqrt((c.x-xp)^2+(c.y-yp)^2);
    return rp;
}

struct stepper_kinematics * __visible
polear_stepper_alloc(char axis)
{
    struct stepper_kinematics *sk = malloc(sizeof(*sk));
    memset(sk, 0, sizeof(*sk));
    if (axis == 'x') {
        sk->calc_position_cb = polear_stepper_rn_calc_position;
        sk->active_flags = AF_X;
    } else if (axis == 'y') {
        sk->calc_position_cb = polear_stepper_rp_calc_position;
        sk->active_flags = AF_Y;  
    }
    return sk;
}
```
But when i want to connect with my printer the octopi 's terminal write this: "There was a timeout while trying to connect to the printer".

If i don't include the kin_polear.c to the source_files my printer connects and runs normaly.

Sineos, 2022-03-11:
It is preferred to post such question in the developer section of https://klipper.discourse.group/
Might also increase your chance to get an answer from the dev.

Sineos, 2022-03-11:
https://klipper.discourse.group/t/how-can-i-add-a-new-file-c-for-my-printer/2269

Thanks. Appreciated

closed 2022-03-11 by Sineos

Marck29, 2022-03-11:
Thanks to you.