Klipper3d/klipper #5348: Defining a second stepper for my Cyclops style hotend results in erroneous error.

UnaClocker, 2022-03-12:
[klippy.log](https://github.com/Klipper3d/klipper/files/8237297/klippy.log)
[printer_cfg.zip](https://github.com/Klipper3d/klipper/files/8237298/printer_cfg.zip)
I have a Cyclops style hotend. Two in, 1 shared heater/nozzle out. The deprecated shared_heater method works. When I try the new method, "[extruder_stepper ...." I get an error message claiming that I'm not specifying the parameter "microsteps" for the additional extruder, even though I have.

Sineos, 2022-03-12:
And as already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/

UnaClocker, 2022-03-12:
This is not a feature request or request for help. It's a BUG!

jalanjarosz, 2022-03-13:
I think "extruder1" is reserved for a motion system. 

I have a discourse issue related to this including a new issue based on the latest release. 

https://klipper.discourse.group/t/switching-between-extruders/2020/7?u=jjarosz

Hope this helps you out, and maybe you will see same issue?  I reverted back a it to keep printing for now.

closed 2022-03-15 by github-actions[bot]