Klipper3d/klipper #5349: High extruder jerkiness on smooth curves

viesturz, 2022-03-12:
I have been noticing that smooth curves have a ton of extrusion jerkiness and unneeded slowdowns.
Note that I'm running a remote direct extruder and it's acceleration is quite limited, which probably makes this issue more obvious.

However there is no reason why such jerkiness should be there for a very smooth curve. It seems that Klipper is decelerating to near 0 and back up for every tiny curve segment.

Especially apparent when tuning pressure advance, likely because SQUARE_CORNER_VELOCITY=1 
In normal operation, it jerks every few hundred ms when the curve radius is small enough. Quite visible on a test spiral. IE https://www.thingiverse.com/thing:4916578

Test shape:  https://drive.google.com/file/d/17u-zKLKbUE2XZSa0oL4FhYM43IYd82FX/view?usp=sharing, 
Test GCode (Cura slicer): https://drive.google.com/file/d/1MROkaKf33Ra9WuDqwHHGpnkwg6Om6piN/view?usp=sharing

Initially I though it was a pressure advance thing, but apparently not, since setting PA to 0 made it even worse.
Crank you volume up to hear the extruder motor being jerked around.
Video of printing with PA=0.1: https://photos.app.goo.gl/MmUk1oQyT9v2Cj8dA
Video of printing with PA=0.0: https://photos.app.goo.gl/UskiCGxNgmv1xGqG8
Video or printing with PA=0.0;  SQUARE_CORNER_VELOCITY=10: Yay, this is what I'm looking for: https://photos.app.goo.gl/XdbGteUPqqGEN8XW7

So high square corner velocity mitigates this, but will likely wreck any sharp corners.

Printer config: [printer.txt](https://github.com/Klipper3d/klipper/files/8237345/printer.txt)
Hardware: Duet wifi + Pi3b + delta kinematics + flex3drive

Sineos, 2022-03-12:
And as already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/

viesturz, 2022-03-12:
> It looks like this ticket is a request for help (or similar). Many helpful people will not see your message here and you are unlikely to get a useful response. Instead, see the contact directions [...]

Bad bot

viesturz, 2022-03-12:
Using a very detailed model helped a bit:
SQUARE_CORNER_VELOCITY=1, with surface deviation of 0.018:  https://photos.app.goo.gl/MmUk1oQyT9v2Cj8dA
SQUARE_CORNER_VELOCITY=1, with surface deviation of 0.003: https://photos.app.goo.gl/vGjHDYkVWzq5Sk4DA
~ 30% cornering speed increase and significantly less jerk.

Starting to dig into the code...

viesturz, 2022-03-12:
So from the docs, the default SCV of 5 is meant to roughly match surface deviation of 0.02. 
And SCV of 1 would translate into ~ 0.0008 which is ridiculously low.
So I suppose this is mostly as intended. Will investigate some more using SCV=5 as reference.

closed 2022-03-14 by github-actions[bot]