Klipper3d/klipper #5351: Bug in PR 5210 - X-In/1-Out Hot End configuration

jalanjarosz, 2022-03-13:
I think there is a bug in PR 5210 / commit v0.10.0-293-ge3beafbd.

I posted an update on Discourse: https://klipper.discourse.group/t/switching-between-extruders/2020/6

SYNC_STEPPER_TO_EXTRUDER has be depreciated for SYNC_EXTRUDER_MOTION, however similar functionality to support X-In/1-Out Extruder/Hot-End is not working when replacing function.

I upgraded switching macro’s to be as such:

```
[gcode_macro T0]
gcode:
    # Deactivate stepper in my_extruder_stepper
    #SYNC_STEPPER_TO_EXTRUDER STEPPER=belted_extruder EXTRUDER=
    SYNC_EXTRUDER_MOTION EXTRUDER=belted_extruder MOTION_QUEUE=

    # Activate stepper in extruder
    #SYNC_STEPPER_TO_EXTRUDER STEPPER=extruder EXTRUDER=extruder
    SYNC_EXTRUDER_MOTION EXTRUDER=extruder MOTION_QUEUE=extruder

[gcode_macro T1]
gcode:
    # Deactivate stepper in extruder
    #SYNC_STEPPER_TO_EXTRUDER STEPPER=extruder EXTRUDER=
    SYNC_EXTRUDER_MOTION EXTRUDER=extruder MOTION_QUEUE=

    # Activate stepper in my_extruder_stepper
    #SYNC_STEPPER_TO_EXTRUDER STEPPER=belted_extruder EXTRUDER=extruder
    SYNC_EXTRUDER_MOTION EXTRUDER=belted_extruder MOTION_QUEUE=extruder
```
When I execute the T0/T1 command with manual extrusion, it operates as expected, however when gcode is run from mainsail, I get this error in the console: `Unable to infer active extruder stepper`

Even if the deprecated commands are used, the same issue exists:

```
13:41:30 Unable to infer active extruder stepper

13:41:29 File selected

13:41:29 File opened:cloak.gcode Size:13052276

13:41:10 Extruder stepper now syncing with 'extruder'

13:41:10 SYNC_STEPPER_TO_EXTRUDER STEPPER=belted_extruder EXTRUDER=extruder

13:40:41 Extruder stepper now syncing with ''

13:40:41 SYNC_STEPPER_TO_EXTRUDER STEPPER=extruder EXTRUDER
```

For manual extrusion commands, it functions, but when initiating motion + extrusion, then Klipper is lost.

I also tried:
`SET_EXTRUDER_ROTATION_DISTANCE EXTRUDER=belted_extruder DISTANCE=0`
but rotation_distance cannot be zero or null.

I reviewed the PR https://github.com/Klipper3d/klipper/pull/5210 and believe I used the intended methodology.

Can you confirm if this is an issue in code or of there is another method that should be used?

jalanjarosz, 2022-03-13:
Extruder configuration:

```
[extruder]
step_pin = PC1
dir_pin = PC3
enable_pin = !PC7
microsteps = 16
rotation_distance = 22.024
full_steps_per_rotation = 200
gear_ratio = 50:17
nozzle_diameter = 0.400
filament_diameter = 1.750
max_extrude_only_distance = 950.0
max_extrude_only_velocity = 250
max_extrude_only_accel = 2750
heater_pin = PB4
sensor_type = ATC Semitec 104GT-2
sensor_pin = PK5
control = pid
pid_kp = 28.114
pid_ki = 1.562
pid_kd = 126.511
min_temp = 0
max_temp = 300

[extruder_stepper belted_extruder]
extruder = extruder
step_pin = PA4
dir_pin = PA6
enable_pin = !PA2
microsteps = 16
rotation_distance = 22.5
full_steps_per_rotation = 200
gear_ratio = 4:1
```

Klipper does not disconnect, but this is the issue in klippy.log:

```
Starting SD card print (position 0)
Unable to infer active extruder stepper
Exiting SD card print (position 108694)
```

full klippy.log is attached in the Discourse link above.

Sineos, 2022-03-13:
It will not help to cross post. Please concentrate your information on discourse. As you know @KevinOConnor is reading there and if needed, I'm sure you will get an answer there.

closed 2022-03-15 by github-actions[bot]