Klipper3d/klipper #5362: Out of bounds aborts (manual) mesh bed leveling

iamcam, 2022-03-19:
Not so much a bug, but a feature request...

When performing manual mesh leveling, if the requested Z position is out of bounds, the mesh bed leveling routine aborts and you must start over from scratch. I understand the reason for this behavior - to prevent the damage to the printer.

Instead of aborting the mesh leveling procedure for requesting an out-of-bounds Z position, it might be a better idea to produce an error message and not perform the move, stating the reason, the requested position, and the boundary position. This would certainly help when performing the calibration points and inadvertently requesting a value only slightly beyond the bounds. I'd rather be told and be able to adjust (maybe I requested too large a move) than have to start over from scratch. This recently happened to me - having to re-start a 15-point leveling procedure a few times over.

Sineos, 2022-03-19:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/

iamcam, 2022-03-19:
Thank you for the direction.

closed 2022-03-19 by iamcam