Klipper3d/klipper #5368: Amount of Z-axis movement is unstable

yhira1023, 2022-03-21:
When I printed in spiral mode, Although the flow rate is constant, the movement amount of the Z axis is uneven. 
As a result, the width of the print changes periodically. 

[image]

the gcode file is made by Cura.
This issue does not occur with marlin on same printer same gcode file.
[TXXPRO_test_c_spiral2.ZIP](https://github.com/Klipper3d/klipper/files/8315059/TXXPRO_test_c_spiral2.ZIP)

As strange point ,Looking at mainsail on printing.
Flow : stable
Z-position : increase unstably

Are there any factors that change the amount of increase in the Z axis? 
or Is it truncated by some calculation? 

My environments
Tronxy X5SA Pro + Orbiter v1.5 
Cura v4.13.1
Klipper /mainsail up to date 
[printer.zip](https://github.com/Klipper3d/klipper/files/8315063/printer.zip)

Sineos, 2022-03-21:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/
Also attach your `klippy.log` there

yhira1023, 2022-03-22:
Okay ,I try support request.
Thank you.

closed 2022-03-24 by github-actions[bot]