Klipper3d/klipper #5370: New EnderPro "Klipper state: Shutdown" while heating up

Laph9979, 2022-03-22:
So I'm setting up a brand new Enderpro 3.  Have printed one small item to verify it worked a few days ago.  Now I'm unable to get it to heat up.  I have gotten it up to temp a few times but then shutsdown.  Before I did the PID It would just fail to keep temp while printing and just shutdown.  After PID I now get the shutdown error.  I have added "max_power: 1.0" and also 

[verify_heater extruder]
heating_gain: 10
check_gain_time: 30
hysteresis: 10
max_error: 120

To attempt to get it to run.  Not sure what I'm missing and hoping someone with more experience in Klipper can figure this out for me.

Also feel free to clean up my config file  ;)
[klippy.log](https://github.com/Klipper3d/klipper/files/8321534/klippy.log)
.

Sineos, 2022-03-22:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/
Also attach your` klippy.log` there

Laph9979, 2022-03-22:
Done, posted in the correct area.  I have included the link for future referance if anyone else has this issue and find this.

https://klipper.discourse.group/t/new-enderpro-klipper-state-shutdown-while-heating-up/2373

Thanks for the referral!

closed 2022-03-22 by Laph9979