Klipper3d/klipper #5371: RPi MCU shutdown due to "Unable write i2c device"

NikosSprocket, 2022-03-22:
A print got aboard due to "Transition to shutdown state: MCU 'Host_rpi' shutdown: Unable write i2c device".
I have installed and using for more than a week now a HTU21D without any isssues. Today, during the print.
[klippy_03-22-022.log](https://github.com/Klipper3d/klipper/files/8327876/klippy_03-22-022.log).
I am running v0.10.0-319-g3e0c0e9c on all MCUs.
I have attached the klippy.log and
[image]

Sineos, 2022-03-23:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/
Also attach your` klippy.log` there

NikosSprocket, 2022-03-27:
> As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/ Also attach your` klippy.log` there

Thank you for recomendation. I did put it in Discourse but after 3 days I have received 0 replies. Anyway, besides development is Github for reporting bugs?

Thanks,
Nick