Klipper3d/klipper #5382: Feature request: extend protocol with UART commands

lanserge, 2022-03-26:
Currently Klipper firmware has support for only one UART with hardcoded use as interface to the Klipper itself.
I have several printers that have external touch screen control modules connected through UART to the main board.
And I can see that it became some kind of industry standard to use such touch screen modules with serial interface connected to the main board. The serial protocol is simple but could be different between modules and versions.
I am thinking of adding support to the Klipper but it needs significant changes in the firmware.
Basically idea is to add several commands to the Klipper protocol to use UART as resource: open UART channel with parameters, write UART channel and receive messages from UART channel.
If someone formalize this idea into generic code for the Klipper then I will be able to implement it for STM32 microcontrollers and may be AVR. With this extension to the protocol it will be easy to support (in python) all possible touchscreen modules or other serial connected modules.

Sineos, 2022-03-27:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/

lanserge, 2022-03-27:
Thank you for pointing out, I will post there.

closed 2022-03-27 by lanserge