Klipper3d/klipper #5386: Allow movement WHENEVER I WANT. Yes, I read #335

gruvin, 2022-03-28:
I just spent a whole day getting Klipper going because everyone raves about it. Now I find that I cannot move anything using standard GCode if the machine doesn't THINK it's homed.

I read the FAQ and all of issue #335. SET_KINEMATIC_POSITION, etc.

$100,000+ machining centers allow themselves to be moved without being homed first. THIS IS STANDARD and SENSIBLE practice. CNC machines all over the place allow setting arbitrary home positions -- in effect telling the machine to DO WHAT I COMMAND regardless.

Software that "protects users from themselves" is HORRIBLE in any engineering context. Just horrid.

Please get rid of this craziness once and forever more.

Now for what it's worth, I predict you're gonna tell me I'm being rude or some such crap and disagree and all that -- so goodbye Klipper. I'm uninstalling it now. What a waste of a day.

Sineos, 2022-03-28:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/

Also I see you are apparently quite frustrated but in the sense of a helpful exchange it makes sense to concentrate on finding solutions instead of blaming.

bhelm, 2022-03-28:
Seeing you being rude against volunteer people offering you a free yet very powerful tool is just sad. Bonus points because you are raging about a missing feature >99,9% of all users will never want to use.

eamaclean, 2022-03-28:
> Now for what it's worth, I predict you're gonna tell me I'm being rude or some such crap

Perhaps in future when you realise you're being rude, you can try not to be. Millions of people around the world do this, it's standard and sensible practice.

gruvin, 2022-03-28:
II know this is offensive but IMO, being offensive is justified when "free" is used to ram bullshit down people's throats. Of course, I am free to just not use the "[offering]" and I'm exercising that freedom. 

Yes, "powerful" it may be but powerful can still be dumb. RIDICULOUS crap like, ">99,9% of all users will never want to use." <== #laughable bullshit. In actually, people GAVE UP requesting it and then the fucking issue was closed for further comment! LOL Are you KIDDING ME? Neo Nazi attitudes like this are abhorrent. Someone has to speak out against it in general across ALL open source development cultures and today, that someone is me.

Presumably, this — my first encounter with Klipper and its "community" is the toxic attitude Tom Salanderer was talking about in his YT video: https://www.youtube.com/watch?v=0E0dM0ZdpRE&t=841s

I didn't even get into Klippers "extended G-code" commands. There's nothing G-code about them! They destroy  the RS-274 standard and demonstrate a complete lack of care for what "99,9%" of manufacturing professionals NEED, letalone want.

gruvin, 2022-03-28:
@[eamaclean](https://github.com/eamaclean) CITATION PLEASE. More bullshit. LOL

gruvin, 2022-03-28:
I'm done. Go back to you Nazi cancel culture now. Bye.

closed 2022-03-28 by gruvin

dalegaard, 2022-03-28:
@gruvin

You seem to think you are entitled to something even though Klipper is provided as-is with no warranty or support. You agreed to this when you downloaded and used the software, it's part of the license agreement.

lf you don't like the software, change it or persuade or pay someone to do it for you. You are not entitled to any time or effort from anyone involved with the Klipper project. This is the power and freedom granted to you by open source.

Calling this neo nazism is not just wrong, it's also deeply offensive and taking a huge crap over all the effort that Kevin and contributors have put into the software over the past many years.

As to your original issue, I think you may need to actually read the discussion you were referring to. Kevin expressed reservations about the feature, but also said that he was willing to incorporate a safe change that handled the various edge cases. No such patch was ever received. The discussion thread even includes a workaround, using `SET_KINEMATIC_POSITION`, which you even note yourself. What is the issue then? That potentially printer-destroying commands are one step removed to protect people from accidentally destroying their printers while messing around with Klipper for the first time? Is that _really_ worth being this rude?

There are toxic people in all communities, but none of the responders to this discussion in any way behaved rude or toxic - in fact, the only one that decided to go that route was you.

Best regards,
- Lasse

gruvin, 2022-03-28:
@dalegaard perhaps if YOU read what I actually wrote, you would see that I read all of that — every word, in fact. I was astonished and frankly disgusted. Dude consciously took away a basic feature that ALL other 3D printer (and CNC) firmware has BECAUSE ITS NEEDED but he refuses to listen, eventually offering an, "oh this might shut them up" hack. SET_KINEMATIC_POSITION indeed. What a joke. Now, I'd not be surprised if the poor fellow is offended by my comments and didn't have these accused intentions. At 51 years old I'm just so over this "cancel history" gen zero crap. We worked hard to create the technical standards and more these people live in and they spit it back in our faces — because you know, "[they] know best". Others were polite. Others tried their best. Others even created pull requests. All were shut down. Those are the facts, there for anyone to see.

OK -- really, this time, I'm gone. Yeah, I know. Good riddens. Whatever.

th33xitus, 2022-03-28:
Im **so glad** you chose to not being part of the Klipper Community. Thank you.

gruvin, 2022-03-28:
> Im **so glad** you chose to not become part of the Klipper Community. Thank you.

You're welcome brother.

mental405, 2022-03-28:
This entire thread was a waste of everyone's time, and after having read it I feel like I have soiled myself.

gruvin, 2022-03-29:
> This entire thread was a waste of everyone's time, and after having read it I feel like I have soiled myself.
Sorry.