Klipper3d/klipper #5392: M108 needed.

gruvin, 2022-04-01:
This is a re-opening of issue #2944, which was IMHO unceremoniously closed without resolution or a care in the world for the OP. (Just saying. shrug)

Therein, @KevinOConnor stated:
> ... The M108 command is broken (as it does not cancel commands sent after M109/M190) and I, personally, am not interested in implementing or maintaining it.

@KevinOConnor that's the entire POINT of `M108`. It is not, "broken". It is a _'break and continue'_, not a _'cancel everything'_. and it is _sorely_ needed by all but the least sophisticated users among us. (It's also used to break from `M0 "message here"` to continue pressing.)

To that end, Is there a Klipper "extended G-Code (not) command" that can do this? If so, we could manually implement `M108` functionally in `printer.cfg`

Thanks.

gruvin, 2022-04-01:
I tried `CANCEL_PRINT` while an `M109 S60` was in progress. That was blocked as the expected behaviour but it also later yielded ...
```
Send: CANCEL_PRINT
Recv: // Unknown command:"CANCEL_PRINT"
```

Maybe that's just a [documentation](url) lag? If so I can issue a PR with it removed. Just say the word.

Sineos, 2022-04-02:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/

gruvin, 2022-04-02:
Confused. Please define “development” in the context of this issue.

This is a bug report, not a feature request. If G28 was missing because of
errant developer opinion — would reporting such critically broken behaviour
be considered a mere feature request?

Feature requests are for non-critical, optional code behaviours. Clearly,
this is not that.

Furthermore, I can implement a fix for this bug myself but will my PR be
accepted when existing developers don’t even understand why it is
critically needed? I think not.

It’s not a feature request when I can code it myself. It’s a debate at
developer level with developers.

Last but not least, since I’m new around here, could you maybe let me know
if I am wasting my time and should give up arguing my point now already.

Thank you.

Sineos, 2022-04-03:
See here for more details: https://www.klipper3d.org/Contact.html

gruvin, 2022-04-03:
OK. read all that. Seems we do not agree on the definitions of "bug", "broken" and "feature request". So, there's no point in continuing this discourse -- anywhere.

Klipper is not for me. #cancelled.

closed 2022-04-03 by gruvin