Klipper3d/klipper #5394: Mainsail-Klipper  configuration problem

soumyajit02, 2022-04-02:
I tried to use my raspberry pi model 3b+ as a secondary Mcu beside the Arduino Mega 2560 which is used as the main control board for the printer followed every instruction described in rpi as a secondary mcu at: https://www.klipper3d.org/RPi_microcontroller.html
changed the printer.cfg according to it as described by the reference document but keeps getting the error...
MCU 'rpi' shutdown: Unable to open out GPIO chip line
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
this is my klippy.log:
[klippy.log](https://github.com/Klipper3d/klipper/files/8403251/klippy.log)

my printer config is ` 
#Mainsail settings
[include mainsail.cfg]

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
   # Run ls /dev/serial/by-id/* for micro-controller name
#restart_method: command
#pin_map: arduino

[mcu rpi]
serial: /tmp/klipper_host_mcu

[stepper_x]
step_pin: ar54
dir_pin: !ar55
enable_pin: !ar38
rotation_distance: 40
Full_Steps_Per_Rotation: 200
microsteps: 32
endstop_pin: ^!ar3
position_endstop: 600
position_max: 600
homing_speed: 50

[stepper_y]
step_pin: ar60
dir_pin: !ar61
enable_pin: !ar56
microsteps: 32
Full_Steps_Per_Rotation: 200
rotation_distance: 40
endstop_pin: ^!ar15
position_endstop: 0
position_max: 480
homing_speed: 50

[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
rotation_distance: 1.25
microsteps: 32
Full_Steps_Per_Rotation: 200
position_endstop: 0.0                     # disable to use proximity sensor
endstop_pin: ^!ar18                        # disable to use Proximity Sensor
#endstop_pin: probe:z_virtual_endstop    # enable to use Proximity Sensor
#position_min: -5                        # enable to use Proximity Sensor
position_max: 340

# [safe_z_home]                         # enable for Proximity Sensor
# home_xy_position: 150, 150
# speed: 100
# z_hop: 10
# z_hop_speed: 5

# [Proximity Sensor]                             # enable for Proximity Sensor - fast-mode
# sensor_pin: ^PB1
# control_pin: PB0
# pin_up_touch_mode_reports_triggered: True
[... 413 lines cut ...]
        line  42:      unnamed       unused   input  active-high
        line  43:      unnamed       unused   input  active-high
        line  44:      unnamed       unused   input  active-high
        line  45:      unnamed       unused   input  active-high
        line  46:      unnamed       unused   input  active-high
        line  47:      unnamed       unused  output  active-high
        line  48:      unnamed       unused   input  active-high
        line  49:      unnamed       unused   input  active-high
        line  50:      unnamed       unused   input  active-high
        line  51:      unnamed       unused   input  active-high
        line  52:      unnamed       unused   input  active-high
        line  53:      unnamed       unused   input  active-high
gpiochip1 - 2 lines:
        line   0:      unnamed       "led0"  output  active-high [used]
        line   1:      unnamed       unused   input  active-high
gpiochip2 - 8 lines:
        line   0:      unnamed       unused  output  active-high
        line   1:      unnamed       unused  output  active-high
        line   2:      unnamed       unused  output  active-high
        line   3:      unnamed       unused  output  active-high
        line   4:      unnamed       unused   input  active-high
        line   5:      unnamed       unused  output  active-high
        line   6:      unnamed       unused  output  active-high
        line   7:      unnamed       "led1"   input  active-high [used]
`

Sineos, 2022-04-02:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/

soumyajit02, 2022-04-03:
> As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/

thanks but tell me where to place this specific issue...

closed 2022-04-05 by github-actions[bot]