Klipper3d/klipper #5408: Heater fan not working as intended

tuncof, 2022-04-11:
Hi,

Configured my heater fan with default options (ie. fan triggers when extruder heater temp. above 50). But when I do an M104 S1 fan triggers regardless of target temperature and/or current temperature. Looking at `klippy/extras/heater_fan.py`,  immediately found the blame:

 **`if target_temp or current_temp > self.heater_temp:`**

As you already know this expands to  **`target_temperature != 0 or current_temp > self.heater_temp`**   So triggers all the time if target temperature but 0.

Very weird that nobody noticed till now despite recent popularity of heater triggered fans. Don't know about recent version changes of python3, so not sure if it's a bug or my rather ancient python (v3.7.3) of raspberry pi zero issue. However paranthesis around the expression miraculously works fine here: 
 **`if ( target_temp or current_temp ) > self.heater_temp:`** 

But of course I have no idea about the original intent. Maybe you mean AND instead of OR?  
**`if target_temp and current_temp > self.heater_temp:`**

Anyways, I'm grateful to you for this quality and clean code both on behalf of me and my happiest printer.
Keep up the good work.

Drachenkaetzchen, 2022-04-20:
Note: Fellow user and Voron supporter here, no dev, so not an official answer.

I think it works as intended.

A heater fan is activated as soon as the associated heater is active. A heater is active when it's temperature is not equal 0. It does **not** wait until the temperature is above 50°C in your example. 50°C is the point where the fan is turned off. 

The use case is rather simple: There's some lag between a heating element and it's thermistor. For a hotend, it makes sense to turn on the hotend cooling fan as soon as the heater starts working, because the thermistor could lag behind.

Sineos, 2022-04-21:
There is maybe a misunderstanding between `heater_fan` and `temperature_fan`.
* Heater Fan: Always active when heater becomes active and turns off when temperature again drops below `heater_temp` (default 50°C)
* Temperature Fan: Temperature controlled fan, where a target temperature can be set that shall not be exceeded (if the fan is able to prevent this, is a different story).

Best would be to continue such a discussion on [Klipper Discourse](https://klipper.discourse.group/) if needed.

tuncof, 2022-04-21:
I know the difference between heater fan and temperature fan. But I know that some people prefers to make nozzle temperature stay at a temporary waiting temperature of say 140 C (which is safe for the PLA material even for prolonged time) and in this use case by this way there is no heating time loss in between two consecutive prints. Machine would stay indefinitely without disturbing sound of fan.  Otherwise does it even make sense to set the trigger temperature? You would simply make it on if any temperature target is set other than zero and off if target is zero without involving the active temperature. Extruder heater fan is needed DURING the print I think not at the end of print. Neither heater nor temperature fan suit for my use case, if I did not get it wrong.

Drachenkaetzchen, 2022-04-21:
@tuncof I think @Sineos was clear that it should be continued on discourse, and that it works as intended.

tuncof, 2022-04-21:
Ok, then developers may think of this as an enhancement request for my (and other people's) fan need?

tuncof, 2022-04-21:
@Drachenkaetzchen: Nope I have a big aluminum cooling block for that matter. But of course during the print if cooler is not active the only problem is clogging due to softened filament.

jakep82, 2022-04-22:
> Ok I misunderstood, then developers may think of this as an enhancement request for my (and other people's) fan need?  

Feature requests also don't belong here. The bot will automatically close this issue in a day or two, so if you want to discuss this request further it needs to happen on Discord or Discourse.

tuncof, 2022-04-22:
Ok, I see. You guys are the ultimate boss. But my final words to argue is that, it would be more reasonable for a scheme that kicks in and shuts off at or near the same temperature (with or without hysteresis). This would broadly encapsulate all the use cases (including mine). And for vast majority users "kicks in at 50 and shuts off at 60" or so would suffice. There are also hardware implentations with MOSFETS/relays around and logic is such. Among them are the famous community member Chuck Hellebuyck's (CHEP) one:  https://youtu.be/NDXUHImA3Xc
So there is no reason to deviate from the common sense. Fundamentally software version should follow the same suit with hardware imho.
Anyways, not a big deal since an easy mod, again thanks a lot for this quality code and keep up the good work dudes.

closed 2022-04-22 by tuncof