Klipper3d/klipper #5411: Unable to reference heater_generic in display code

Makhaira, 2022-04-12:
Klippy log at the bottom of this post

I'm trying to set up a display on my new build utilising configs that are happily working on other printers, and I have a problem that after discussion in the discord #macros channel I believe is a bug.

The problem is being able to reference a 'heater_generic' through display syntax. I get an error that a ',' token was expected. If this uses any other form of heater or temperature sensor then it works without issue. It feels like 'heater_generic' is almost not recognised as something that can be referenced as part of [printer].

[klippy (3).log](https://github.com/Klipper3d/klipper/files/8471740/klippy.3.log)

Sineos, 2022-04-12:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/

Makhaira, 2022-04-12:
Sorry @Sineos but this falls under point 6 in my opinion which is why I posted here. I suggest the Contact page may need updating?

Sineos, 2022-04-12:
Sorry, believing it is a bug, is not knowing. Until verified, please revert to discourse.

Makhaira, 2022-04-12:
ok, I'll close this. Can I suggest the guidance is updated please? I feel like I've wasted people's time despite reading the guidance.

closed 2022-04-12 by Makhaira

Sineos, 2022-04-12:
> Be sure the bug is in the Klipper software. If you are thinking "there is a problem, I can't figure out why, and therefore it is a Klipper bug", then do not open a github issue. In that case, someone interested and able will need to first research and diagnose the root cause of the problem.

I think this is pretty clear. Feel free to suggest a better wording.