Klipper3d/klipper #5422: Batch set_led commands failing when final transmit=1 command is unchanged

mstansberry, 2022-04-15:
Issue:
Commit 8b8818c0f7c12cc97b2f1658c3305683e41ec42d appears to have changed the `set_led` command batch behavior preventing batch `set_led` commands from applying when the final command including `transmit=1` does not include updated values. 

Testing:
Reverted to 96795def9c90276bcf3ea00d2292002237363ed6 and behavior is as expected

Steps to reproduce:
Set values for multiple indexes on a chain with the final `transmit=1` value applying changes, then update a single `transmit=0` index color values and re-run command. Changes will not be applied until a color value is updated for the index which includes `transmit=1`

julianschill, 2022-04-17:
This is probably the line causingbthis issue: https://github.com/Klipper3d/klipper/blob/44567879f668ec414a445b8fbbe50806ff6b6b10/klippy/extras/led.py#L50

If the color is not changed the function returns without transmitting. But it should continue, because there cozld be changes from prevous SET_LED commands. So we either need some "dirty" bit or we send data in any case. I would prefer sending it, because there could also be the intention to refresh the LEDs with the unchanged colors.

mstansberry, 2022-04-18:
Just verified, this is now working as expected. Thanks so much for the quick turnaround.

closed 2022-04-18 by mstansberry