Klipper3d/klipper #5423: "printer.pause_resume.is_paused == true" after print has finished

masterxq, 2022-04-15:
Hi *,
after finish a print is finished the pause state is sometimes true
```
printer.pause_resume.is_paused == true
```

**Story**
The aim is to issue a shutdown command that only shutdowns the printer if no print is active or paused.

**Expected behavior**
If a print has paused the state is true else should be false.

**Actual behavior**
Sometimes the state is true after a Print has finished.

**Steps to reproduce**
The problem only occurs sometimes, and I haven't been able to determine exactly how to trigger it. The issue may appear more often if the print is paused more often or the idle timeout is triggered.
I will attach additional observations if available.

The attached log should contain a Print that caused the issue (the second to the last) and another that did not cause the issue (the last).
I will attach additional logs if available.

[klippy.log](https://github.com/Klipper3d/klipper/files/8497944/klippy.log)

**Additional question**
What is the best way to detect if an incomplete print is active? My current strategy is:
```
if printer.idle_timeout.state == "Idle" AND printer.pause_resume.is_paused == True
    shutdown
```

Thank you for reading.

Sineos, 2022-04-17:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: [Klipper Discourse](https://klipper.discourse.group/)

masterxq, 2022-04-18:
Ok, sorry, it was not supposed to be a feature request or support request. The Story was just for give context and tell why this could be important for users. I removed the question from the report. I hope is acceptable now.
I absolutely understand that this report can not be confirmed like it is, as there are missing information. But if other users have the same issue the data can be collected here and the problem sharpened.
I absolutely see, I should not be that lazy add a question here, but the rest should be a quite conventional report for an issue.

Additional I fully understand the objective to have as few open issues as possible. To archive this AND allow people to collect data to corresponding problems and issues it could be a really good idea to use tags or something equivalent. It feels absolutely bad if some has make the work to collect an issue as good and precisely as possible and it will be rejected by convention. Collecting information about problem is an important part of a development progress!
It is not possible to archive flawless reports in an open source project, this is even a hard task in a small team where people know each other.
Klipper is a really nice project. Many of the people writing here are developer and the only really active developer is KevinOConnor and i understand why, if some should use his/her sparetime for a opensource project, it needs to feel good. This is feels not good. Change the tone and the policy and the core developer will have time to breathe again.  
Additional, I found multiply ok fine bug reports where i had additional information to track the problem. All of them was closed by convention and i had no option to add my data. In my opinion this is really a bad case and takes the power of a opensource project where a good QA could be free!

I'm writing this because I really totally understand why it is like it is, but it should not if you want to have a team somewhen. People with power will not accept this, only the dummies just here for solving there own problems will do so!

As my report now it is fixed and we should not steal other people time with reading this and as we should not discus this here too, feel free to delete all post to this issue (except the report itself ofc).
But please take a second and think about my words. 

Thanks for understanding my position

Arksine, 2022-04-18:
If you pause a print without doing one of the following the state will not reset:
1) Call the `RESUME`
2) Call `CLEAR_PAUSE`

It is wise to add `CLEAR_PAUSE` in your start gcode, as it will guaranteed that a stale paused state is cleared before you resume.

closed 2022-04-20 by github-actions[bot]