Klipper3d/klipper #5424: STM32F103 delivers wrong SPI speed

cooked, 2022-04-16:
A control board based on the STM32F103 delivers the wrong SPI speed. The speed looks like to be 1/2 of the configured one 

When default SPI parameters are used, for example to communicate with a TMC, the speed should be 4M, while 2M (lin picture, ittle deviation is due to the signal/scope framing)  is measured on the clock line. 
The issue might be related to the fact that the default Klipper SPI, SPI2, is on APB1 where the maximum running frequency is 36MHz and not 72MHz  

[image]
[image]
[image]

KevinOConnor, 2022-04-17:
As far as I know, the code does what it is intended to do.  They key code is in src/stm32/spi.c:
```c
    // Calculate CR1 register
    uint32_t pclk = get_pclock_frequency((uint32_t)spi);
    uint32_t div = 0;
    while ((pclk >> (div + 1)) > rate && div < 7)
        div++;
    uint32_t cr1 = ((mode << SPI_CR1_CPHA_Pos) | (div << SPI_CR1_BR_Pos)
                    | SPI_CR1_SPE | SPI_CR1_MSTR | SPI_CR1_SSM | SPI_CR1_SSI);
```

The `get_pclock_frequency()` function returns 36000000 on stm32f103 (see stm32f1.c for details).  That loop will result in `div=3` which is a division of16 and 36000000 / 16 == 2250000 frequency.  That's the highest frequency that the spi clock can be programmed to without exceeding the requested 4Mhz clock.

-Kevin

cooked, 2022-04-17:
Thanks for clarifying. The reference does not describe that and is somewhat vague in regard 
[image]

Sineos, 2022-04-25:
@cooked If the topic is concluded, please be so kind to close the issue.

closed 2022-04-25 by cooked