Klipper3d/klipper #5427: SKR Mini E3 V1.2 Breaks USB detection after V0.9.0

CNCmarlin, 2022-04-18:
Hello longtime marline user here, really want to try out klipper but I have spent 10+ hours on getting my SKR mini e3 to function! 

I have attached a log of my skr using klipper v0.1.0. When using the firmware made from this version of klipper with and without swd enabled (Im almost positive I have a SMC board on not a GD version). There is a part in the log near the very bottom where it sees my board but the is from a firmware I found online from v0.9.0 with this firmware my board is recognized by the pi but klipper does not like it they are not form the same version.

More information that may be helpful

while using the firmware flashed from 0.1.0 Nothing registers in dmesg when plugging/unplugging the USB same with lsusb command.
when issuing ~/klipper $ ls /dev/serial/by-id/* I receive: cannot access ‘/dev/serial/by-id/*’: No such file or directory
Klipper log just complains about invalid MCU.

[image]

I cannot for the life of me understand what I may be doing wrong to cause my board to not work properly except when using the firmware from v0.9.0. Kind Regards

P.S. I really think this is a cool looking piece of software you guys are creating!

[klippy.log](https://github.com/Klipper3d/klipper/files/8502962/klippy.log)

CNCmarlin, 2022-04-18:
I forgot to mention this the version of the board is 1.2

Sineos, 2022-04-18:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: [Klipper Discourse](https://klipper.discourse.group/)

closed 2022-04-18 by CNCmarlin