Klipper3d/klipper #5433: Spurious/unsuppressable "pin ADC_TEMPERATURE used multiple times in config" error

GarthSnyder, 2022-04-20:
I originally had a single ARM controller board with an internal temperature sensor (BTT Octopus). I wanted to drive the case fans according to the temperature reported by the MCU, but I also wanted the MCU temperature displayed in mainsail. So, I referenced the sensor in both a `temperature_fan` for the case and a separate `temperature_sensor` clause. Klipper complained about the duplication. I then added a `duplicate_pin_override` for ADC_TEMPERATURE and everything was good. So far, perfectly reasonable.

Now I've added a separate Easy BRD board to control an Enraged Rabbit Carrot Feeder. It's based on a SAM D21 chip, so it has an internal temperature sensor. Klipper is once again complaining about the duplicate ADC_TEMPERATURE pin, even if I don't make any explicit reference to the D21's internal sensor. (But of course, I want the Easy BRD's temperature displayed in mainsail, just as for the Octopus. I've tried defining a separate `temperature_sensor`, but it doesn't seem to make any difference.)

There are, perhaps, two different issues here:

* The Easy BRD sensor is not the same pin as the Octopus sensor, even in a virtual sense. They are completely different and shouldn't cause Klipper to think there's a collision. All explicit references are fully qualified by MCU (via `sensor_mcu`).

* There's already a `duplicate_pin_override` for ADC_TEMPERATURE. Why is Klipper ignoring it now? The docs seems to suggest that `duplicate_pin_override` should allow more than two references to a pin.

Abbreviated configuration:

```
[duplicate_pin_override]
pins:
    ADC_TEMPERATURE

[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32f446xx_1D0008000F5053424E363620-if00
restart_method: command

[temperature_sensor octopus]
sensor_type: temperature_mcu
sensor_mcu: mcu

[temperature_fan electronics]
pin: PD12
...
sensor_type: temperature_mcu
sensor_mcu: mcu
...

[mcu ercf]
serial: /dev/serial/by-id/usb-Klipper_samd21g18a_395871963352555020312E36432319FF-if00

[temperature_sensor ERCF]
sensor_type: temperature_mcu
sensor_mcu: ercf
```
[klippy.log](https://github.com/Klipper3d/klipper/files/8517421/klippy.log)

jakep82, 2022-04-20:
Help requests should go on Discourse, but to give you a quick answer you appear to have a typo in your config leading to a parsing error.  There is no duplicate_pin_override for ADC_TEMPERATURE.  Instead I see the lines below in your log.

```
[virtual_sdcard]
path = /home/pi/gcode_files
	ADC_TEMPERATURE
```

GarthSnyder, 2022-04-20:
Yep, total pilot error - my apologies. The Easy BRD configuration adds a `duplicate_pin_override` of its own that shadows the original.

closed 2022-04-20 by GarthSnyder